Ultimaker/Cura #5378: cura4 beta2 GUI error

UltiArjan, 2019-03-01:
**Application Version**
cura4 beta2

**Platform**
WIN10

**Printer**
S5, but like the same on others

**Steps to Reproduce**
**best look at the enclosed video, it explains the best.**
When you move a object on the build plate to the right you can't position it all the way to the right part of the build plate when the setting window is next to it.

**Expected results**
be able to move the object anywhere on the visible part of the buyildplate , not being blocked by other parts of the GUI

**Additional Information**
I would really like the settings window be be "dockable" to maximise the right side of the screen, and not interfere with area containing the print buttons (bottom right)

UltiArjan, 2019-03-01:
[4Beta2-gui.zip](https://github.com/Ultimaker/Cura/files/2920983/4Beta2-gui.zip)

fieldOfView, 2019-03-01:
See the Sidebar GUI plugin in the Marketplace. It fixes this problem by docking the settings window as you request.

UltiArjan, 2019-03-01:
I feel what shows in the video is a bug and needs to be fixed, its a separate issue I think

I know about the plugin... but the docking should also be a default option imho, but if not ill definitely use your plugin :)  so thanks!!!

fieldOfView, 2019-03-02:
I see. Cura 4 with my plugin exhibits the same issue, but so does Cura 3.6 (and likely all versions before it). It just became a bit more reproducable in Cura 4 because the settings floater extends more into the viewport.

For developers: see the video in the 1st comment. What seems to happen is that the mouse coordinate used to move the object makes a horizontal jump to the left when the mouse is dragged over the settings floater (or sidebar in previous versions) while repositioning the model. This only happens with the settings floater; the toolbar and tool panel don't do this.

Ghostkeeper, 2019-03-04:
Yes, one of our testers found this issue as well. It takes the position where it should move to depending on the mouse coordinates, but for some reason the mouse coordinates are taken relative to the settings panel when you hover the settings panel (instead of properly relative to the main Cura window).