Ultimaker/Cura #5380: [Cura 4.0.0] - Wrong first layer temperature

romualdrichard, 2019-03-02:
**Application Version**
4.0.0

**Platform**
Windows 10

**Printer**
Ender 3

**Steps to Reproduce**
Having 
; Ender 3 Custom Start G-code
M104 S210 ; Set Extruder temperature
M140 S60 ; Set Heat Bed temperature

In the printer parameter

Setting 70 as temperature for the bed

**Actual Results**
Temperature of the bed is going to 70 then, print is stared and temperature goes to 60 during the first layer, after, goes up again

**Expected results**
Always 70 for the first layer or the custom code should use the temperature of the project

**Additional Information**

romualdrichard, 2019-03-02:
Removing M104 and M140 is solving the issue but temperature is changed by this setting till second layer

Ghostkeeper, 2019-03-04:
You set this 60C temperature in the start g-code, so of course it's going to heat up to that temperature then. Cura recognises your use of M140 and omits its own M140 command in the start-up then. Because Cura has both an initial layer bed temperature setting and an ordinary bed temperature, it'll change the temperature to the ordinary bed temperature after the first layer is completed. This is all expected behaviour.

What you should do instead, is set your start g-code to this:

    ; Ender 3 Custom Start G-code
    M104 S{material_print_temperature_layer_0}
    M140 S{material_bed_temperature_layer_0}

And also, instead of M104 and M140 you might want to use M109 and M190 to also cause the printer to wait until it's heated up. But that depends on the firmware too so maybe it won't be necessary for you.

closed 2019-03-04 by Ghostkeeper