Ultimaker/Cura #5381: wanhao duplicator i3 no mods , not connect to cura ultimaker 3,6 Suggestion ?

miguelcnc, 2019-03-02:
i connect my wanhao duplicator i3 , standard , without important modifes , only goood alignement of belt and bed , 

i install the 3.6 versio n of ultimaker cura , but i can't connect the sw to my printer, with other sw ( matter control ) work fine ... 

any suggestion for me ? i reinstall win 7 64 clean and i ( crazy.... ) upgrade all ms updates, but nothing change , the sw ask me  " not connected to the printer ks to the comunity for help ."   any idea for solve this issue ? 

thanks  , Stefano , Fron ITALY

Ghostkeeper, 2019-03-04:
You're connecting via USB cable?

Maybe MatterControl can tell you what Baud rate it is using in order to connect?

miguelcnc, 2019-03-04:
Hi , yes i hVe connected via usb cable , the speed of ports are 115200 , i try also to update drivers for the interface converter and also for usb port , but nothing change , if i use other sw ( matter control ) the Machine print.... I Begin to be crazy ...The cable is good mini usb ended long 1 mt only , the os is win 7 64 , i reinstall also from dvd and all drivers are ( i suppose ) correct , i tri with 2 notebook  and same issue...Inviato da smartphone Samsung Galaxy.
-------- Messaggio originale --------Da: Ghostkeeper <notifications@github.com> Data: 04/03/19  16:59  (GMT+01:00) A: Ultimaker/Cura <Cura@noreply.github.com> Cc: miguelcnc <miguelcnc@alice.it>, Author <author@noreply.github.com> Oggetto: Re: [Ultimaker/Cura] wanhao duplicator i3 no mods , not connect to cura ultimaker 3,6 Suggestion ? (#5381) You're connecting via USB cable?
Maybe MatterControl can tell you what Baud rate it is using in order to connect?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Ultimaker/Cura/issues/5381#issuecomment-469306189",
"url": "https://github.com/Ultimaker/Cura/issues/5381#issuecomment-469306189",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Ghostkeeper, 2019-03-05:
115200 seems normal. It's the first baud rate that Cura will try, even.
Since your printer connected fine with MatterControl but not with Cura, the fault seems to be in Cura.

However I have no way to debug this since I don't have a Wanhao Duplicator 1 here. I can't test anything.

miguelcnc, 2019-03-06:
are you developer of cura team ? if you like , i can also permit you to connect to my pc with teamviewer , so you can check if all is in order ... 

Thanks for now, if think matter control , use proprietary drivers to connect to the printer, but the melzi board  use the ftdi drivers and is updated drivers... i begin to  be crazy -... 
 thanks fo rnow .

  ----Messaggio originale----

 
 Data: 5-mar-2019 17.16
 

 
 
 

 

 
 Since your printer connected fine with MatterControl but not with Cura, the fault seems to be in Cura. Not with any cables, drivers or OS.

Ghostkeeper, 2019-03-07:
Yes, I'm one of the [Cura developers](https://github.com/Ultimaker/Cura/graphs/contributors). But I think that even in a remote desktop application I couldn't help a whole lot. I would check the baud rates and if it responds to `M105` g-code, but that's all I could really do.

miguelcnc, 2019-03-07:
i check  in the later afternoon ... and i reply you . Thsanks for now . 
Stefano

  ----Messaggio originale----

 
 Data: 7-mar-2019 9.59

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-09 by github-actions[bot]