Ultimaker/Cura #5383: Cura 3.6.1 - Tool Change including move to (0,0)

XBrav, 2019-03-03:
I have my dual extruder printer set up with Marlin 2.0.x with nozzle offsets pre-programmed, so offsets in the slicer are not required.

When slicing the dual colour benchy, Layer:0 finalizes with a G0 F3600 X0.00 Y0.00, which crashes the hotend out of bounds. An additional G0 is issued to set the Z height for Layer:1, but also includes a X0.00 and Y0.00. I cannot find a setting for this, and it is not included in the tool change scripts. It only happens on the first layer transition.
[image]
[CuraDualExtruder-HomeMovements.zip](https://github.com/Ultimaker/Cura/files/2922779/CuraDualExtruder-HomeMovements.zip)

smartavionics, 2019-03-03:
As a workaround set Layer Start X/Y to a position nearer to the middle (i.e. 80/80).

smartavionics, 2019-03-03:
You have the setting prime_blob_enable set to true and I think that is causing the travel back to 0,0 to prime that extruder. Doing the prime on the 2nd layer has to be a bug because the extruder has already just been used to print the outer few lines of the brim.

smartavionics, 2019-03-03:
As an alternative to setting the layer start x/y you can also set extruder_prime_pos_x (and y).

XBrav, 2019-03-03:
Interestingly enough, the setting seems to be auto enabled for Custom FFF Printer by default, but is hidden and therefore not setable / unsetable:

[image]

smartavionics, 2019-03-03:
I find the whole Cura settings mechanism totally bewildering, I don't think I will ever understand how it works because there's no definitive documentation that one can learn from.

Ghostkeeper, 2019-03-05:
The bug isn't caused by any setting I don't think. When slicing that project file in the original post, my debug build of CuraEngine is giving an assertion failure:
```
2019-03-05 17:48:18,841 - DEBUG - [Thread-23] UM.Backend.Backend._backendLog [101]: [Backend] [WARNING] Warning: Couldn't find previous extruder plan so as to set the standby temperature. Inserting temp command in earliest available layer.
2019-03-05 17:48:18,842 - DEBUG - [Thread-23] UM.Backend.Backend._backendLog [101]: [Backend] [WARNING] Empty extruder plan detected! Discarding extrusion temperature command.
2019-03-05 17:48:18,843 - DEBUG - [Thread-23] UM.Backend.Backend._backendLog [101]: [Backend] CuraEngine: /home/ruben/Projects/CuraEngine/src/LayerPlanBuffer.cpp:89: void cura::LayerPlanBuffer::addConnectingTravelMove(cura::LayerPlan*, const cura::LayerPlan*): Assertion `newest_layer->extruder_plans.front().extruder_nr == prev_layer->extruder_plans.back().extruder_nr' failed.
2019-03-05 17:48:19,413 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [811]: Backend quit with return code -6. Resetting process and socket.
```
So this is more of a slicer bug. I'll post this up for my colleagues to discuss.

The "Enable Prime Blob" setting is only made visible for Ultimaker 3 and Ultimaker S5 since it causes Cura to insert a g-code `G280` that was developed in-house for the UM3. Other printers won't support that (and if they do, their printer definition file could enable this feature I guess).

XBrav, 2019-03-05:
So, the start X and Y positions did work to prevent the crash, but it still seems like an unnecessary move.

Thanks for looking into it @Ghostkeeper

Vandrasc, 2019-03-08:
Fixed for 4.0.

Ghostkeeper, 2019-03-11:
The prime blob setting visibility issue was fixed, but the move to 0,0 was not.

Block137, 2019-07-05:
I found that my printer will try to move to Layer Start X/Y position after tool change.
Default is 0,0, I changed it to 400,400 and it crashed to upper-right corner.

This is in 4.1

A simple search and replace script such as from "X0." to ";X0." can workaround this problem.

rburema, 2019-10-08:
It might be good to mention that the relevant internal ticket is CURA-6279

3DStyle, 2020-03-12:
Seems like I'm experiencing this issue in Cura 4.5.0. Did this bug got fixed?

fieldOfView, 2020-03-13:
> Did this bug got fixed?

Since the issue is still open, probably not.

rburema, 2020-03-17:
Sorry, that bug is indeed still in the todo column.

matthuszagh, 2020-11-27:
I'm experiencing the same issue in 4.8.

RFBomb, 2021-01-22:
Just sliced in 4.8.0 and I also had the nozzle issue unexpectedly moving to approximate XY 0. 

[image]

Vandresc, 2021-04-15:
This should be fixed inow n the next 4.9 release. Please give it a try when the release is made public. 

Thank you for your feedback!

closed 2021-04-15 by Vandresc