Ultimaker/Cura #5384: Skirt is printed after the Tree Support during first layer

jairov4, 2019-03-03:
**Application Version**
3.6.0

**Platform**
Windows

**Printer**
TEVO Tarantula

**Steps to Reproduce**
Disable Support
Enable Tree Support (In experimental section)
Enable Skirt
Enable Support brim*
Print!

**Actual Results**
The Skirt is printed after the tree support during first layer

**Expected results**
The first thing to print should be the Skirt

**Additional Information**
Please fix this, Im very happy with the tree support results, its just a simple quirk sometimes preventing the perfect printing

rburema, 2019-03-05:
Hi @jairov4,
I've tried to reproduce this in 4.0-BETA-2, and found it printed skirt first, even when I switched around the adhesion and/or skirt/brim extruder a couple of times. Could you try that version? The 'final' 4.0.0 will follow within a few weeks.
cheers, Remco.

jairov4, 2019-03-05:
sure I will try it :)

MightyCreak, 2019-03-24:
I do have the same issue (on an Anycubic model), but not only with tree support. I'll share you some screenshots.

Screenshot of the issue w/ tree support:
[image]
As you can see, it's starting the skirt after the tree supports, which is highly problematic (at least with my printer) since it takes a good 50 mm until it starts extruding, so basically the first one or two supports are missing their base, and never adhere to the build plate with the next layers.

Screenshot of the issue w/o tree support:
[image]

Here are the 3MF files: [AKLP_support_tests.zip](https://github.com/Ultimaker/Cura/files/3000606/AKLP_support_tests.zip)

MightyCreak, 2019-03-24:
Here is the result of the print with tree support:
[image]

rburema, 2019-03-26:
@MightyCreak I don't have the printer definition you have 'anycubic_kossel_plus', as such, I couldn't reproduce it (that is, I can only get skin befir the rest is printed). But the missing definition only allows me to import the model; could you maybe send the definition as well?

rburema, 2019-03-26:
... scratch that, I can see it now: it needs 'enable support brim' set to true. Which makes it a duplicate of  ticket #4829. We're working on that! We know what cuases it, and both internal tickets will very probably end up in 4.1

closed 2019-03-26 by rburema

MightyCreak, 2019-03-26:
Actually, "Enable Support Brim" is already set to true.
[image]

rburema, 2019-03-26:
@MightyCreak Yes, I needed that option on 'true' to reproduce the bug.

MightyCreak, 2019-03-26:
I'll wait for the next update then ;)
Thank you!