Ultimaker/Cura #5385: [3.6.0] Missing retraction on travel within wall leads to under extrusion on outer face

bytte1, 2019-03-03:
**Application Version**

3.6.0
**Platform**

Win 7
**Printer**

Anet A8
**Steps to Reproduce**

Slice project (in attached zip - together with photo and diagram) with default settings, 0.15 layer height, support everywhere, 100% infil, retraction enabled. The project is an instrument box. [image]
**Actual Results**

When printed there is under extrusion on the outside of the box [image]
This is likely due to travel without retraction for at least the length of the side of the box in the WALL:OUTER as shown in the diagram, derived from the generated gcode for an affected layer.
[image]
**Expected results**

The printed box will have the defects shown (printer is not doing hardware retraction)
**Additional Information**

[Bug.zip](https://github.com/Ultimaker/Cura/files/2923361/Bug.zip)

smartavionics, 2019-03-03:
Set combing to "no skin" and set the max comb distance without retract to something like 10.

bytte1, 2019-03-03:
That appears to have solved the original problem, but the lower layers of the walls are now 'stuttering' due to tiny steps  
[image]
whereas the top layers are normal.
[image]

smartavionics, 2019-03-03:
Please save the project file, zip it up and attach to this issue, thanks.

bytte1, 2019-03-04:
[PI3_BodyFacedDown_stutters.zip](https://github.com/Ultimaker/Cura/files/2923856/PI3_BodyFacedDown_stutters.zip)

bytte1, 2019-03-04:
Looks like that was 'caused' by the infill density having been at default when I loaded the project file. Setting infill to 100% avoids the issue, although I suggest that it ought not require infill to be 100% to satisfactorily print this. 

Analysing my actions, I now recognise that I had not saved the project file at the time that the original gcode was generated. The project file was created from the same .stl file as the original gcode, but  I had not changed the non-default settings.  I then reloaded that project file for the second print and neglected to override the defaults for brim to skirt, infill to 100%, temp to 220 and generates support everywhere.

smartavionics, 2019-03-04:
Thanks for the file. Unfortunately, that isn't a Cura project file, it only contains the model and not the settings. Please do File -> Save rather than File -> Export.

bytte1, 2019-03-04:
[PI3_dds v5 v9_FaceDownProjectFile.zip](https://github.com/Ultimaker/Cura/files/2925711/PI3_dds.v5.v9_FaceDownProjectFile.zip)

smartavionics, 2019-03-04:
Hello @bytte1 , thanks for the project file. That short zigzag pattern is being created by the fill gaps between walls option. I cannot explain why the top layers do not show the same pattern. I know that the Cura devs are working to improve the gap filling so you may want to try the 4.0 beta release. You can also try one of my releases which do the gap filling differently and so do not produce the zigzags. You can find my releases at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

mporter88, 2019-03-09:
"Set combing to "no skin" and set the max comb distance without retract to something like 10."

The above instruction worked for me.
Upgraded CURA version to 3.6.0 and decided to tune in from defaults.
Was missing all retractions after the first.
CR-10.

Thanks to Smart Avionics.

eskeyaar, 2019-12-23:
@smartavionics I have query, what does combing "all" does? I searched many post regarding good documentation about combing all but couldn't find any., I have same similar issue when I turn on combing all with max comb distance without retract. If you can lead me to proper example of combing all vs combing to no skin would be helpful. Thanks

smartavionics, 2019-12-24:
Hi @eskeyaar, combing all enables combed travel moves everywhere within a part's outline, it can travel over the infill, skin and inner walls. No skin combing removes the part's skin areas so the travel can no longer cross the skin. Finally, the old infill only variant only combs within the part's infill areas but, unfortunately, does not care how it routes between infill so it will happily cross regions of skin. Personally, I nearly always use the no skin setting. I rarely use the all setting and never use the infill setting. And sometimes, i.e. when printing TPU, I don't use combing at all. Hope this helps.

Ghostkeeper, 2019-12-24:
Here is a write-up of the four options in the Combing drop-down in my Settings Guide plug-in: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/travel/retraction_combing.md

eskeyaar, 2019-12-24:
Thanks @smartavionics and @Ghostkeeper

closed 2020-06-22 by nallath