Ultimaker/Cura #5386: BUG: send to octoprint(offline) gets queued somehow

tablatronix, 2019-03-03:
**Application Version**
3.6.0

**Platform**
osx

**Printer**
MPSMV2

**Steps to Reproduce**
* in octoprint, disconnect printer.
* in cura, send job to octoprint, warning says is cannot, close dialog
* in octoprint connect printer
* in cura send to octoprint, jobs gets sent
* octoprint starts printing

It seems that failed send jobs still gets saved, cached somewhere,  and then the next job gets appended to it.

**Actual Results**
BUG
* octoprint starts printing the FIRST failed job!
* octoprint finishes job, but says 50%
* octoprint starts printing the second job!

**Expected results**

Not doing that thing..

**Additional Information**
Also what does the `queue` button do, cura nor octoprint have a queue afaik..
Maybe closing the box is doing something wrong with this queue?
I have not tried clicking it to see if behavior changes or not.

fieldOfView, 2019-03-04:
Please check if there’s a newer version of the OctoPrint Connection plugin available in the Marketplace. This issue was fixed two releases ago.

tablatronix, 2019-03-05:
I had no idea what marketplace was, I kept looking for plugins, extensions, doh
updated, will test again

closed 2019-03-05 by tablatronix