Ultimaker/Cura #5387: Add Combing "Not in first layer" (Ooze issues)

mat2k, 2019-03-03:
Combing "Not in Skin" doesn't seem to consider the first layer "skin" at all, even though it is? (correct me if i'm wrong, please). This makes it so that my printer will travel without retracting when making the first layer walls, leaving lines on the bare glass bed. When the first layer skin is made these lines stay as imperfections and give it a really bad finish. The solution to this is disabling combing altogether, which makes some prints excruciatingly slow.

I believe a fix for this would be either adding a checkbox to "Disable Combing in first layer" or actually treating the first layer skin as skin in "Not in Skin" mode.

I'm using Cura 3.6.0 on Win10, my printer is a custom prusa i3 rework-based printer

smartavionics, 2019-03-03:
Hello @mat2k , please save your Cura project file that has this problem, zip it up and attach to this issue. Thanks.

mat2k, 2019-03-03:
this happens on literally any file i use, and I don't believe it's inherent to my Cura install or printer config (or STL files for that matter). I'll upload a random file, though. But I believe this can be reproduced easily simply by slicing any file and checking the first layer travels - on "Not in Skin" mode, there are no light blue travels when the printer has to travel from the wall finish to the skin start.

Is this expected behavior? Is the first layer skin actual skin?

smartavionics, 2019-03-03:
> Is this expected behavior? Is the first layer skin actual skin?

Until I see your project file I can't comment on the behaviour but, yes, the first layer skin really is skin and the no skin combing mode should avoid it.

mat2k, 2019-03-03:
actually, i should set my minimum travel length with retract to a lower value. i'll close this before it gets more attention 😆

closed 2019-03-03 by mat2k