Ultimaker/Cura #5389: Cura not slicing entire object

jruckman91, 2019-03-04:
When slicing in Cura, it only slices the last couple of layers and not the whole object. I've reinstalled the program, adjusted settings, tried to repair the object but nothing is working. I was able to print everything just fine but it is doing this with every .stl file.
[image]

smartavionics, 2019-03-04:
Hello @jruckman91 , please save a project file, zip and attach to this issue. Thanks.

jruckman91, 2019-03-04:
[tabletop 1.zip](https://github.com/Ultimaker/Cura/files/2927618/tabletop.1.zip)

jruckman91, 2019-03-04:
Hopefully I sent it correctly.

smartavionics, 2019-03-04:
I'm afraid not. You sent the model but I want the whole project (that includes the model and the settings you are using). So do File->Save and that should produce a .3mf file which you can zip and attach.

smartavionics, 2019-03-04:
That's File->Save from within Cura!

jruckman91, 2019-03-04:
[AI3M_tabletop 1.zip](https://github.com/Ultimaker/Cura/files/2927632/AI3M_tabletop.1.zip)

smartavionics, 2019-03-04:
it slices fine for me and looking at your image above it looks like it has sliced OK for you. The lower layers are greyed out but they can be displayed by moving the layer slider down and up again (not sure why it has started greying out the lower layers. Perhaps that's a bug?)

jruckman91, 2019-03-04:
It slices in cura and shows that it did just fine. But when you go to layer view and use the slider next to the play button it only shows it only printing that last layer or so. Then when I save it to print it only prints the last layer or 2 that it showed in cura. Hopefully that makes sense.

smartavionics, 2019-03-04:
I saved the gcode and looked at it in a gcode viewer and it shows all the layers so I can't see there's a problem.

[image]

jruckman91, 2019-03-04:
Do you think its some kind of bug in Cura? I am unable to print anything. Even things I have printed previously are not being sliced correctly and only have the last layer printed. I will send you another example.

jruckman91, 2019-03-04:
[image]
If you look closley it only shows the last layer being printed. If I were to save this and print it out, only the yellow portion would be printed.

smartavionics, 2019-03-04:
Please send me both the project file and the gcode that you get from it so I can slice it myself using the same settings and compare the gcode I get with what you are getting. Thanks.

jruckman91, 2019-03-04:
[tabletop legs.zip](https://github.com/Ultimaker/Cura/files/2927828/tabletop.legs.zip)
Here is the gcode for the legs. The profile is the same as the one I have sent previously.

jruckman91, 2019-03-04:
I am about to leave for work so I will not be able to send any more information. But I can comment from my phone. Thank you for your help!

smartavionics, 2019-03-04:
LOL, that wasn't gcode, that was an image! Don't worry, please send the STL and the gcode when you can.

jruckman91, 2019-03-04:
Oh oops haha. My bad, I will send you it once I get home tonight. Thank you!

jruckman91, 2019-03-05:
[table_legs.zip](https://github.com/Ultimaker/Cura/files/2929249/table_legs.zip)

jruckman91, 2019-03-05:
hopefully I did it correctly now haha

smartavionics, 2019-03-05:
Thanks for the table legs model. Well, I can slice this and the gcode looks OK so I really don't know where the problem is here.

jruckman91, 2019-03-05:
I just don't understand what is going on. I've had my 3d printer for about a year and haven't changed anything as far as the way I print. This issue just happened the other day. I must have some kind of bug. I will try to reinstall it again. Maybe an older version. Thank you for all your help.

jruckman91, 2019-03-05:
Still no luck. It does the same thing no matter what version I download.

Ghostkeeper, 2019-03-05:
> (not sure why it has started greying out the lower layers. Perhaps that's a bug?)

Yes, something to do with how it detects that it should be greyed out and the initial state after a reslice ending up like that.

> I just don't understand what is going on. I've had my 3d printer for about a year and haven't changed anything as far as the way I print. This issue just happened the other day. I must have some kind of bug. I will try to reinstall it again. Maybe an older version. Thank you for all your help.

Maybe it would help to compare the g-code that Cura generates now with the g-code that Cura used to generate in versions that did work. Maybe some g-code is causing it to skip towards the last layer?

jruckman91, 2019-03-05:
I've had Cura 3.6 and its been slicing just fine and my prints are great. Then all of the sudden this started happening two days ago. It does this with every g-code no matter what version of Cura I use. I even tried other slicer software and I get the same issue.

Ghostkeeper, 2019-03-05:
If it happens with all slicers then you need to take a hard look at the firmware. Perhaps do a factory reset?

jruckman91, 2019-03-05:
A factory reset on my pc?

Ghostkeeper, 2019-03-05:
No, on your printer.

jruckman91, 2019-03-05:
Its not the printer that is the problem. Its the slicer software. I don't have my printer hooked up to my pc. I save the g-codes to a sd card and then load them up in my printer. When I am in Cura, I load in a stl file, slice it and then when I check it in layer view it only shows the last layer being printed.

jruckman91, 2019-03-05:
[image]
Here is another example. If I were to save this to my sd card and have my printer print it, it will only print the yellow area.

jruckman91, 2019-03-05:
No settings were changed. I've used the same settings for the past year. Only things changed are layer height and infill percentage.

smartavionics, 2019-03-05:
> Here is another example. If I were to save this to my sd card and have my printer print it, it will only print the yellow area.

Are you saying that it would print that yellow area on the print bed (i.e. z = 0) or that it would try to print it way up in the air (and fail, of course).

Ghostkeeper, 2019-03-05:
> I even tried other slicer software and I get the same issue.

So you're saying that all slicer software has the exact same bug, that it only prints the top-most layer?

jruckman91, 2019-03-05:
For me yes, I've tried two other slicers and I get the same issues. The picture I sent I have printed before and printed just fine. But it will print directly on the bed only the yellow area now.

smartavionics, 2019-03-05:
> For me yes, I've tried two other slicers and I get the same issues. The picture I sent I have printed before and printed just fine. But it will print directly on the bed only the yellow area now.

OK, well if it's occurring with gcode from other slicers also then it can't be a problem that is specific to Cura. Your printer must be having a bad hair day!

jruckman91, 2019-03-05:
I haven't tried printing from the other slicers I used because when I preview the layers and travels it only shows that last layer. So I haven't even attempted to print anything anymore because it just fails. It can't be a problem with my printer because its not even involved in this issue. It's when I load in the stl file to slice it. thats when the problem occurs.

smartavionics, 2019-03-05:
OK, I'm really confused now. So you're not actually printing anything, you're just viewing the gcode in Cura?

BTW, Just because the layers are shown as grey in the layer view doesn't mean that they won't be printed.

jruckman91, 2019-03-05:
This is exactly what happened. I loaded in a stl file into Cura. Sliced it, saved it to my sd card and went to print it. The print failed and only printed the last layer. So I checked it in the layer viewer in Cura and just like the failed print, it only showed that last layer. I closed Cura, opened it back up to slice it again. The same file sliced perfectly and it showed all the layers that would be printed. I saved it to my sd card to print and it printed perfectly. But now every time I load in a file to slice within Cura it only shows the last layer. I havent tried to print anything since then. I've only been printing what is on my sd card because I have orders that need to be printed. Hopefully this makes sense. Sorry for any confusion. I'm trying to explain it the best way I can.

Ghostkeeper, 2019-03-05:
Here's a thought then: Maybe the file on the SD card was corrupted? Was the SD card ejected before removing it from the computer?

jruckman91, 2019-03-05:
Yes I eject it every time.

jruckman91, 2019-03-05:
Its possible that file got corrupted but why would it not show me every layer and travel when I slice a file in Cura? It has previously.

Predator75, 2019-10-09:
I have just installed the last version today and I have the same problem, Cura slice only the last layer

blkexp98, 2020-01-01:
I am having the same issue.  I dont even get to the SD card portion and use Octoprint anyway.  This is a new computer and I havent tried to print again until now.   Also when I login to octoprint I can load and simulate an old file which works fine (within octoprint) but the new ones I have tried all do what the original poster described.   So this has to be something with how the slicer is working.  The only other variable would be I was using an older intel based computer and now I built  AMD Ryzen based PC.

Ghostkeeper, 2020-01-01:
I think your issue is different then Blkexp98, since the original report says that this happens with every slicer software.

blkexp98, 2020-01-01:
Do you want me to start a new thread so its easier to keep track of?

Ghostkeeper, 2020-01-02:
It's probably best to start a different thread about it. Be sure to include some reproduce steps and maybe a project file. Loading that calibration cube in an Ultimaker 3 printer for me makes it print fine, though some of the letters are significantly thinner than 0.4mm and won't get printed then.

closed 2020-06-22 by nallath