Ultimaker/Cura #5390: 3.6 Issue - Pause and Resume results in Extruder unloading non-stop

n9jcv, 2019-03-05:
I sliced 2 models with Cura 3.6.  I have a Hyper Cube running a chinese SBASE smoothie board clone.  I was attempting to change the filament mid print to have a multi colored object.  

After 20 layers, I pressed the pause on the TFT screen, the printer stopped and the bed dropped roughly 20mm.  Then I removed the filament, fed in a new color, then I clicked the Resume on the TFT.  At this point, the bed came back up and the filament reversed from the extruder, leaving no filament.  The extruder would not stop turning.  

I tried this a 2nd time thinking I had done something wrong.  Same result.  I previously used Cura 15 and the same steps in 15 work perfectly.  Hardware was not changed, just slicer.  Firmware and config on board was not changed.

Please let me know if I can offer other items for debug purposes.
Thank You

Ghostkeeper, 2019-03-05:
The difference has to be something in the g-code that the two slicers generate. Maybe one is extruding in relative mode and the other is extruding in absolute mode?

n9jcv, 2019-03-06:
I am away from home now and cant investigate

closed 2020-06-20 by no-response[bot]