Ultimaker/Cura #5391: Cura 3.6 Slicing Error

novio8, 2019-03-05:
I printed Delta Calibration tool and the print has a Geometry error
as seen from Picture of Repetier Preview (Red Arrow pointing at error)
Cura shows Preview as it should be sliced.
I attached  Repetier Preview

Cura 3.6
Mac OSX 10.13.6
Kossel Delta
[image]

novio8, 2019-03-05:
I used the same Cura 3.6 profile in Cura 2.5.1
and at firs glance Model is sliced correctly, so there must
be something going on with 3.6

smartavionics, 2019-03-05:
See https://community.ultimaker.com/topic/26878-cura-36-slicing-bug/

Ghostkeeper, 2019-03-05:
I'm also not able to reproduce that, using the model that's posted on the forum :disappointed:

novio8, 2019-03-06:
sorry for delay, I solved the problem by the following procedure
1. saved my Cura profiles and materials
2. removed  all Cura installations
3. Reinstalled Cura 3,6
4. Imported back my Profiles
 
Now it seems everything is just fine

Ghostkeeper, 2019-03-07:
Something in your configuration was awry I guess, but I suppose we'll never find it out now. I'm glad you found a workaround for it.

closed 2019-03-07 by Ghostkeeper

ediciuStsaF, 2019-03-09:
[image]

TOP doesn´t slide correctly

for me new install doesn´t work....

[image]

Ghostkeeper, 2019-03-11:
In your case it looks like the top surface there is exactly on the border between two layers. What happens if you change the initial layer height to be 0.01mm higher?

It's hard to say whether the model is at fault here or the slicer.

ediciuStsaF, 2019-03-11:
In Slic3r this Model slices correct. I‘ll give it a try bei changing 0,1mm