Ultimaker/Cura #5392: [4.0.0-PPA] Segmentation fault when choosing preview tab

thopiekar, 2019-03-05:
**Application Version**
4.0-branch - installed via PPA

**Platform**
Ubuntu Linux

**Printer**
- Doesn't matter -

**Steps to Reproduce**
Clicking on the preview tab

**Actual Results**
Boom [1]

**Expected results**
Able to get into the preview view and simulating the print.

**Additional Information**
[1]
```
[MainThread] UM.Resources.__initializeStoragePaths [414]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [425]: Config storage path is /home/thopiekar/.config/cura/4.0
[MainThread] UM.Resources.__initializeStoragePaths [433]: Data storage path is /home/thopiekar/.local/share/cura/4.0
[MainThread] UM.Resources.__initializeStoragePaths [445]: Cache storage path is /home/thopiekar/.cache/cura/4.0
[MainThread] UM.PackageManager.__init__ [53]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager.__init__ [53]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [138]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [145]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [149]: Initializing version upgrade manager ...
[MainThread] UM.Settings.ContainerRegistry.addContainer [384]: Container [empty_definition_changes] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [384]: Container [empty_variant] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [384]: Container [empty_material] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [384]: Container [empty_quality] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [384]: Container [empty_quality_changes] added.
[MainThread] UM.PackageManager._loadManagementData [87]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [87]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [104]: Loaded user packages management file from /home/thopiekar/.local/share/cura/4.0/packages.json
[MainThread] UM.PackageManager._saveManagementData [172]: Package management file /home/thopiekar/.local/share/cura/4.0/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [90]: Loading plugin configuration file '/home/thopiekar/.config/cura/4.0/plugins.json'
2019-03-05 10:14:03,506 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ConsoleLogger
2019-03-05 10:14:03,532 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [81]: Found CuraEngine at: /usr/bin/CuraEngine
2019-03-05 10:14:03,532 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraEngineBackend
2019-03-05 10:14:03,533 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ConsoleLogger was already loaded
2019-03-05 10:14:03,535 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-03-05 10:14:03,539 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-03-05 10:14:03,542 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-03-05 10:14:03,544 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-03-05 10:14:03,547 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-03-05 10:14:03,551 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-03-05 10:14:03,552 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-03-05 10:14:03,554 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-03-05 10:14:03,556 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJWriter
2019-03-05 10:14:03,559 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLWriter
2019-03-05 10:14:03,561 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJReader
2019-03-05 10:14:03,565 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLReader
2019-03-05 10:14:03,568 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-03-05 10:14:03,570 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimpleView
2019-03-05 10:14:03,571 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [372]: Plugin [CuraBlenderPlugin] has been disabled. Skip loading it.
2019-03-05 10:14:03,573 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [372]: Plugin [CuraOpenSCADPlugin] has been disabled. Skip loading it.
2019-03-05 10:14:03,576 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [372]: Plugin [CuraInventorPlugin] has been disabled. Skip loading it.
2019-03-05 10:14:03,579 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [372]: Plugin [CuraFreeCADPlugin] has been disabled. Skip loading it.
2019-03-05 10:14:03,584 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [372]: Plugin [CuraSolidWorksPlugin] has been disabled. Skip loading it.
2019-03-05 10:14:03,596 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPWriter
2019-03-05 10:14:03,603 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdateChecker
2019-03-05 10:14:03,609 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin XRayView
2019-03-05 10:14:03,615 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraProfileReader
2019-03-05 10:14:03,623 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ChangeLogPlugin
[... 516 lines cut ...]
  File "/usr/lib/python3.6/threading.py", line 426 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f81c23ee740 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/ControllerProxy.py", line 108 in _onActiveViewChanged
  File "/usr/lib/python3/dist-packages/UM/Signal.py", line 315 in __performEmit
  File "/usr/lib/python3/dist-packages/UM/Signal.py", line 212 in emit
  File "/usr/lib/python3/dist-packages/UM/Controller.py", line 102 in setActiveView
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/ControllerProxy.py", line 40 in setActiveView
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/ControllerProxy.py", line 105 in _onActiveStageChanged
  File "/usr/lib/python3/dist-packages/UM/Signal.py", line 315 in __performEmit
  File "/usr/lib/python3/dist-packages/UM/Signal.py", line 212 in emit
  File "/usr/lib/python3/dist-packages/UM/Controller.py", line 166 in setActiveStage
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/ControllerProxy.py", line 48 in setActiveStage
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/MainWindow.py", line 168 in mouseReleaseEvent
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 326 in exec_
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 770 in run
  File "/usr/bin/cura", line 133 in <module>
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

Also affects someone else, so shouldn't be related to my computer too much: https://github.com/thopiekar/Cura-packaging/issues/64

fieldOfView, 2019-03-05:
Does the AppImage work on the same computer?

thopiekar, 2019-03-05:
Checked the beta 2 now.. Interestingly no.
Something I also notice is that the margins are missing in some buttons in my packages.
[image]

But that must be another story.

fieldOfView, 2019-03-05:
That last bit reminded me of this issue, which is worth a check: https://github.com/Ultimaker/Cura/issues/5307

rburema, 2019-03-05:
We're still working on some issues w.r.t. the PPA for 4.0 final.

@fieldOfView, the issue you just mentioned was eventually fixed by 'don't rely on HTML, use Qt as what it's meant for instead', I doubt that's causing the wrong margin in those buttons :-)

fieldOfView, 2019-03-05:
Ah. The last posts in that issue seemed to blame a newer Qt version.

rburema, 2019-03-05:
Oh I see. But in that case I guess that shouldn't matter for the AppImage?

Ghostkeeper, 2019-03-05:
Both the segfault when loading the QML and the missing margins could be caused by having a different Qt version.

thopiekar, 2019-03-05:
Are you really sure it is about Qt?
I mean during the last years Cura together with different Qt versions was never so fragile.
I understand that a minimum version is needed for the new theme, but I remember that I used Cura about 1...1.5 months before and it worked fine on the same machine.
Sadly I recently use Cura with my PPA too less, so I don't notice when these problems happen.

Ghostkeeper, 2019-03-05:
No, I'm not sure this is Qt, but it seems a likely candidate if the Qt version is indeed different.

We did see some breaking of backwards compatibility with Qt 5.11 for some people (though for me it works fine).

thopiekar, 2019-03-05:
Any experiences with 5.12?

Btw. noticed that I'm on Qt 5.11.1, but there is 5.11.3 on Debian already. Normally Ubuntu is pushing hard in updates. Wondering why they stopped here.
However don't see any specific patches between 5.11.1 and 5.11.3[1]. If there is something, then it must have been provided by Qt already.

--------------------
[1] https://metadata.ftp-master.debian.org/changelogs//main/q/qtbase-opensource-src/qtbase-opensource-src_5.11.3+dfsg-5_changelog

Ghostkeeper, 2019-03-06:
As far as I know, nobody in this office tried 5.12 yet.

snostormjp, 2019-04-06:
I am having the same issues on arch linux with cura 4.0.0, will build from git and retry
Qt 5.12.2 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 8.2.1 20181127) on "xcb" 
OS: Arch Linux [linux version 5.0.5-arch1-1-ARCH]

thopiekar, 2019-04-06:
Great! Thank you for your efforts in doing investigations around this! 👍

snostormjp, 2019-04-06:
okay , so compiled master from git and had issue on my 2head desktop where it would not go past splash screen , different issue, so went over to my laptop and compiled from git there and am having same segfault for preview.   

Relevant Arch linux build script for cura-git

url="https://github.com/Ultimaker/Cura"
provides=('cura')
conflicts=('cura')
depends=('python'
         'python-pyqt5'
         'python-numpy'
         'python-protobuf'
         'qt5-graphicaleffects'
         'qt5-quickcontrols'
         'qt5-quickcontrols2'
         'libarcus-git'
         'libsavitar'
         'uranium-git'
         'curaengine-git')
makedepends=('git' 'cmake')
optdepends=('python-zeroconf: Detecting mDNS printers'
            'cura-binary-data-git: Firmwares and translations'
            'python-pyserial: USB printing'
            'cura-fdm-materials-git: Default Materials')
source=('git+https://github.com/Ultimaker/Cura.git')
md5sums=('SKIP')

pkgver() {
  cd Cura
  git describe --tags | sed 's/-/./g'
}

build() {
  mkdir -p Cura/build
  cd Cura/build
  cmake -DCMAKE_INSTALL_PREFIX=/usr -DURANIUM_SCRIPTS_DIR=/usr/share/uranium/scripts ..
  make
}

package() {
  cd "$srcdir/Cura/build"
  make DESTDIR="$pkgdir/" install
}

thopiekar, 2019-04-07:
Hmm, I'm not an ArchLinux expert, but I don't see where you are using Qt 5.12+ in your script. 🤔

snostormjp, 2019-04-07:
qt 5.12 is current on arch linux, the script was the build requirements for
compiling from git.  I was trying to start debuging but it is not very
friendly to python -n trace --trace /bin/cura  which crashed before I get
to even the splash screen,

thopiekar, 2019-04-07:
Maybe using gdb beneath python? Sounds like the crash is happening before Python's debugger is able to come up.

Ghostkeeper, 2019-04-11:
I'm currently using Qt 5.12 via the PyQt installation on Pip (Ubuntu 18.10) and running Cura fine.

thopiekar, 2019-05-04:
Fixed the problem in my PPA.

closed 2019-05-04 by thopiekar

snostormjp, 2019-05-06:
curious as to how you fixed the ppa

thopiekar, 2019-05-06:
Is there something you want to know?