Ultimaker/Cura #5395: "Minimum Diameter" (Support towers) should be called "Maximum Diameter"

AndreKR, 2019-03-06:
The tooltip for Support -> Minimum Diameter says:

> Minimum diameter in the X/Y directions of a small area which is to be supported by a specialized support tower.

This ("minimum") sound like an overhang has to have **at least** this diameter to get a support tower.

[The docs](https://ultimaker.com/en/resources/20422-cura-support-settings) make it a bit clearer, but still call it a "minimum diameter":

> Support towers are only printed for support areas that are **smaller** than the set minimum diameter.

(Emphasis mine.)

Shouldn't it be the "**maximum** diameter of an area to still be supported by a support tower"?

Ghostkeeper, 2019-03-06:
Yeah, I think you're right. This is where the setting is being used:

https://github.com/Ultimaker/CuraEngine/blob/24e57cbd6b215ca21b4cfb31ca474dec600042a9/src/support.cpp#L966

I think we'd have to rename this one. In general "Minimum Diameter" was a bit of a weird name as it did not specify that it belongs to support towers, which is okay if reading the settings in the list but not if you're filtering settings via the search box.

LipuFei, 2019-03-08:
JIRA ticket: CURA-6311

closed 2019-06-17 by pkuiper-ultimaker

Ghostkeeper, 2019-06-18:
The setting has been renamed to Maximum Tower-Supported Diameter now, which is more accurate and more specific in that it mentions towers.

PR #5888 (2019-06-17): Rename Minimum Diameter to Maximum Tower-Supported Diameter
The new setting name describes better what the setting is meant to convey.

Implements CURA-6311 and fixes #5395.