Ultimaker/Cura #5403: [FEATURE REQUEST] allow different thickness (Z) for outerwall

JLMorange, 2019-03-07:
this looks like the https://github.com/Ultimaker/Cura/issues/4840
but a little bit different....

what I would like is to perform the inner print with a thickness that is a multiple of outer to save printing time 

for example I want to print the outer wall at 0.1mm for surface and the internal at 0.3 for strength and time saving, enabling this option would print 3 layers at 0.1 with only the outerwall and it's projection, then after ending the last layer of outer wall will print the inside will extruding 0.3*head_diameter (eg print at 0.3 layer thickness) 
same for top/bottom structure

Ghostkeeper, 2019-03-07:
This exists for infill right now. Try the setting Infill Layer Thickness. It does exactly this.

It does not exist for top/bottom. I think that would reduce the quality of the top/bottom quite significantly when you attempt that: More holes in your skin, worse sagging and worse bed adhesion.

GregValiant, 2024-11-24:
I'm cleaning up here.
It does not appear that this will be implemented.
Can we close this?

closed 2024-12-08 by github-actions[bot]