Ultimaker/Cura #5404: [FEATURE REQUEST]  allow export of print parameter and printer settings

JLMorange, 2019-03-07:
**Application Version**
3.6

**Platform**
windows

**Printer**
multiple

**Steps to Reproduce**
either : 

- create a print profile (changing layer height, print speed, or any settings that are not printer related), export it, change the printer, try to import the profile...
- create a custom printer, with custom gcode initialization, platter size... no button to export the printer settings in order to install it on another computer/user
 

**Actual Results**
currently, there is no way to export the printer settings alone and when exporting the print profile, there is printer settings included so you can't import the print profile into the other printer, so you need to reconfigure all by hand (or copy the whole cura settings....)

**Expected results**
have a "export printer" and "import printer" into the printer settings
have the export print profile not include the printer settings

**Additional Information**

Ghostkeeper, 2019-03-07:
You can export/import by clicking on File -> Save. Then load the file it produces on a different computer.

JLMorange, 2019-03-11:
I want to import/export PRINTERS and PRINT PROFILES not models..... I want to be able 
1) to allow the users to import the printer profile they want to print on and add them to their printers
2) allow the user to import some print material-print thickness profile and add them to their print profile

Ghostkeeper, 2019-03-13:
Yes, so clear your build plate, then click File -> Save. This stores the printer as well as the quality profile ("print material-print thickness profile", in your words) and any custom profile.

Alternatively, if you want to export only the custom profile, go to "Manage Profiles" under the profile selection drop-down and click Export. This won't export the custom printer settings though.

JLMorange, 2019-03-28:
I tried to clear the build plate, then save the file and open it on another account, this didn't created my custom printer

I tried to copy the whole environment (in appdata->local->cura) from one user to another, this didn't recreated my custom printer.... so right now, I can't share my custom printer between users....
thanks and regards

Ghostkeeper, 2019-03-29:
Hold on, are you using Custom FFF printer or did you mod Cura with your own .def.json files?

JLMorange, 2019-03-29:
I created a custom fff printer

Ghostkeeper, 2019-04-01:
I cannot reproduce this issue then. Here are the steps I'm using to try to reproduce it:

1. Add a Custom FFF Printer.
2. Give the printer some weird dimensions to recognise it easier (200x10mm build plate).
3. Save a project file.
4. Close Cura.
5. Reset Cura's configuration (delete `~/.local/share/cura`, `~/.config/cura` and `~/.cache/cura`).
6. Start Cura again.
7. Go through the welcome set-up, adding a completely different printer (not Custom FFF) just to be sure.
8. Load the project file, as a project file not just the models.

I then see the Custom FFF Printer again with the customised dimensions.

JLMorange, 2019-04-03:
ok I could import the printer using your process, the issue was that I was importing the models only.... 
thanks and sorry for the reporting of an allready available feature (might be nice to have a button in the printer panel to export an empty project and one to import the project...)
thanks and regards

closed 2019-04-03 by JLMorange