Ultimaker/Cura #5406: Smooth Spiralized Contours Behavior Persists, Despite De-selection

calinb7, 2019-03-08:
**Application Version**

3.6.0
**Platform**

MXLinux18 64-bit (Debian based) using the Cura Appimage
**Printer**

Ultimaker Original
**Steps to Reproduce**

1. Select "Spiralize Outer Contour."
2. Process a file.
3. Select "Smooth Spiralized Contours" checkbox.
4. Process a file.
5. De-select "Smooth Spiralized Contours" checkbox.
6. Process a file.

**Actual Results**

"Smoothing" behavior (x/y jitter) persists.

**Expected results**

Smoothing behavior should cease. (Results of step 6 and step 2, above, should be the same.)

**Additional Information**

smartavionics, 2019-03-08:
Hello @calinb7 , please zip up some example gcode with and without the smoothing enabled and attach to this issue so that I can investigate. Thanks.

calinb7, 2019-03-08:
Hi Mark,

I tried to duplicate the bug but I cannot do so.

However, the behavior that I'm seeing now is not right either! The "Smooth Spiralized Contours" checkbox does not result in smoothing at all now.

When it did "work" (and I could not turn it off), I could see the "jitter"/ smoothing in the Cura layer view graphic as well as in my UMO steppers and print (looked like jitter). Now I see no jitter in the Cura layer view graphic.

Here's how I made the attached files.

First I made the 

1. no_smooth_selected.gcode file.

then I checked the smoothing box and made

2. smooth_selected.gcode file

and finally I unchecked the smoothing box and made

3. smooth_de-selected.gcode file.

No other changes were made. I used linux "diff" to compare the results and got new surprises. :(

Here's the diff between the first two files (there is no difference in the actual gcode even though one file had smoothing selected):

$ diff '/home/demo/Downloads/Resizeable_snap-together_box/smooth_selected.gcode' '/home/demo/Downloads/Resizeable_snap-together_box/no_smooth_selected.gcode'

4205,4206c4205,4206
< ;SETTING_3 5\\nmagic_spiralize = True\\nsmooth_spiralized_contours = True\\n\\n"
< ;SETTING_3 }
---
> ;SETTING_3 5\\nmagic_spiralize = True\\nsmooth_spiralized_contours = False\\n\\n
> ;SETTING_3 "}

I also attached a dff file for the diff between 2nd and 3rd file, where there is actually a difference, but there should be NO difference, because I returned to the original settings when the made the third file (no_smooth_selected.gcode and smooth_de-selected.gcode had the same Cura GUI settings).

The behavior is different than what I reported in my bug report but there's still apparently a bug here.

Best Regards,

-Cal

--------------------------------------------

 

 
 it on GitHub, or mute
 the thread.

smartavionics, 2019-03-08:
Hello Cal. Unfortunately, you can't attach files in an email response to github. You have to come here to the web page and attach them in a reply.

calinb7, 2019-03-08:
Here are the files.

[smoothing_debug.zip](https://github.com/Ultimaker/Cura/files/2945016/smoothing_debug.zip)

closed 2019-03-08 by calinb7

smartavionics, 2019-03-08:
Thanks for the files. I see you have closed the issue but here's my thoughts anyway:

1 - there's no difference between the smoothed and non-smoothed gcodes because the model has a cross section that doesn't vary with z.

2 - the gcode that is different is due to the fact that the model has moved on the build plate or something else has caused the coordinates to differ?

Bye!

calinb7, 2019-03-08:
Hmm--I'll let you know if I see the sticky jitter behavior again.