Ultimaker/Cura #5409: Fan speed

lwbsbde, 2019-03-09:


rburema, 2019-03-10:
Hi @lwbsbde , 
Could you attach the project file for which this happens? (File > Save... then rename to zip to circumvent githubs' restrictions.)

Ghostkeeper, 2019-03-11:
Possibly this is because the model abruptly goes from a large layer to a medium-sized layer?

lwbsbde, 2019-03-11:
I have slice an Easter egg Model with different settings.
1. regular speed 30 , max 100, fan speed threshold 13s and minimum layer time 2s
2. the same but minimum layer time 4s
3. minimum layer time 8s

Layer longer than  13 s ,it is always 30 %
Layer shorter than 13 s , it is 
Setting 1: at least 50% speed
Setting 2: at least 63% speed
Setting 3: at least 80% speed

The minimum layer time causes the speed jump

lwbsbde, 2019-03-11:
> Possibly this is because the model abruptly goes from a large layer to a medium-sized layer?

No , the layer time changes slowly.

lwbsbde, 2019-03-11:
> Hi @lwbsbde ,
> Could you attach the project file for which this happens? (File > Save... then rename to zip to circumvent githubs' restrictions.)
[PATTERN_-_KRAKEN.zip](https://github.com/Ultimaker/Cura/files/2952727/PATTERN_-_KRAKEN.zip)

lwbsbde, 2019-03-11:
> Hi @lwbsbde ,
> Could you attach the project file for which this happens? (File > Save... then rename to zip to circumvent githubs' restrictions.)
> PATTERN_-_KRAKEN.zip

lwbsbde, 2019-03-11:
> Hi @lwbsbde ,
> Could you attach the project file for which this happens? (File > Save... then rename to zip to circumvent githubs' restrictions.)

that happens with all objects

lwbsbde, 2019-03-11:
With Cura 4.0.0 beta , I have the same problem

Ghostkeeper, 2019-03-13:
It's a bit easier to see with just a tetrahedron. With your settings, I see the result that you're wondering about. It seems that when the preliminary estimate becomes less than Regular/Maximum Fan Speed Threshold, it makes a jump from 30% to 77.6%. And then interpolates between 77.6% and 100% depending on how close it comes to the minimum layer time.

I don't know why that happens though, because in the code it seems to interpolate normally between regular and maximum fan speed:

https://github.com/Ultimaker/CuraEngine/blob/99ee78237b8d356ccc6096a2575fa4d73e15d166/src/LayerPlan.cpp#L1280-L1284

lwbsbde, 2019-03-13:
> It's a bit easier to see with just a tetrahedron. With your settings, I see the result that you're wondering about. It seems that when the preliminary estimate becomes less than Regular/Maximum Fan [...]

if the minimum layer time 0, than are the time between regular and the maximum 13s
if the minimum layer time 4, than are the time between  only 9s . But the code interpolare it as if it were 13s. And cut it to the fan speed threshold

lwbsbde, 2019-03-13:
I mean the error lies in the calculation of the layer time difference

lwbsbde, 2019-03-14:
[Star_big_0075_Spike_10x.stl.zip](https://github.com/Ultimaker/Cura/files/2966194/Star_big_0075_Spike_10x.stl.zip)
A tetrahedron project file.

Ghostkeeper, 2019-03-14:
> 1282 : what is "fan_speed_layer_time_settings.cool_min_layer_time_fan_speed_max" ?
is it the threshold time ?

Yes, that's the setting "Regular/Maximum Fan Speed Threshold" for the currently active extruder.

> I think "fan_speed_layer_time_settings.cool_min_layer_time_fan_speed_MIN" is better in the place !!
> See line 1256

`cool_min_layer_time_fan_speed_MIN` doesn't exist. The ASCII art is a bit unclear. In time, it's interpolating between `cool_min_layer_time_fan_speed_max` and `cool_min_layer_time`.

lwbsbde, 2019-03-14:
does it work with you when you slice an object with minimum Layer time?

Ghostkeeper, 2019-03-15:
Once it's below the minimum layer time it works fine for me. It's between that and the Regular/Maximum Fan Speed Threshold that it seems to make a mistake on my end.

These calculations are in CuraEngine.exe on your computer (if you're using Windows; otherwise just CuraEngine). You can't easily edit them unless you know how to compile CuraEngine.

lwbsbde, 2019-03-15:
I could not translate your answer clearly. 
Can you confirm the mistake as he does me ?

Ghostkeeper, 2019-03-19:
I'll keep it simple for your translation tool then :yum: 

> Can you confirm the mistake as he does me ?

Yes.

> Is it a bug in the Programm?

Yes.

> Maybe he does not subtract in line 1282 .

Maybe. I don't know.

> Can I preview the layer time and fan speed in the preview of cura?

No. You can preview the printing speed (feedrate). That should allow you to see the minimum layer time, but not the fan speed.

userosos, 2019-04-28:
Hi! This problem actual for Cura 4.0.
The piramide for tests.

[CalibrationPyramid.zip](https://github.com/Ultimaker/Cura/files/3125309/CalibrationPyramid.zip)
Need scale to ~40 mm the sizes.
regular/maximum fan speed threshold - the parametr not work.

rburema, 2019-05-07:
@Devs: Do we have a ticket for this yet?

Ghostkeeper, 2019-05-07:
@rburema Yes, CURA-6410.

lwbsbde, 2019-05-27:
did you find the error in the program?

Ghostkeeper, 2019-05-28:
Not yet. It's slowly crawling towards the top of our todo list though.

lwbsbde, 2019-09-27:
did you now find it

Ghostkeeper, 2019-09-30:
Not yet. It's still in our bug backlog.

BarsMonster, 2020-01-03:
I also face this issue at the moment.

clfaye, 2020-03-02:
I have this same issue printing a small pyramid:
regular fan speed = 20%
maximum fan speed = 32%
Regular/Maximum Speed Threshold = 1
Regular Fan Speed at Layer = 15
Minimum Layer time = 5

Cura ignores the regular fan speed.  The fan speed increases from 0 to 32% at layer 15.  Layer 15 takes 5 seconds, so the fans should only be at 20%

lwbsbde, 2020-05-01:
Hallo ...
could  anyone  kill the bug
PLEACE

BarsMonster, 2023-01-09:
> I also face this issue at the moment in 4.4.1

I still see this in 5.2.1

BarsMonster, 2023-01-09:
LayerPlan.cpp function processFanSpeedAndMinimalLayerTime lines 1666-1670:
Layer time is compared to configured fan speed which is probably why this is happening. 

```
    else if (fan_speed_layer_time_settings.cool_min_layer_time >= fan_speed_layer_time_settings.cool_min_layer_time_fan_speed_max)
    {
        fan_speed = fan_speed_layer_time_settings.cool_fan_speed_min;
    }
    else if (force_minimal_layer_time && totalLayerTime < fan_speed_layer_time_settings.cool_min_layer_time_fan_speed_max)
```

wawanbreton, 2024-07-02:
Fixed in https://github.com/Ultimaker/CuraEngine/pull/2111
Sorry for the quite long delay :sweat_smile:

MariMakes, 2024-07-16:
👋 Quick update on our side.
This is resolved in the 5.8 release 🎉 , you can download the Beta version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.8.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.8.0-beta.1)

Thanks again, and please let us know if you run into any other issues 💪