Ultimaker/Cura #5410: ChangeatZ Postprocessing not working when print via USB is used.

bastion-0078, 2019-03-09:
Software:  Cura 3.60

OS: Windows 10

Printer: Anycube I3 Mega (stock) via USB

Problem:

Using Postprocessing (ChangeatZ) to change Layer-Temperature. Then I run the job via USB but the Printer prints the whole object with the starting Temperarure.

Workaround:
Saving the post-processed object as G-Code file, print it via SD-Card or open it in Cura and print the presliced G-Code file will work. The printer changes the Temperature at specified layers.

rburema, 2019-03-10:
Hi @bastion-0078 ,
Strange, it should be exactly the same gCode, so the printer should behave the same. Just to check: You (re)sliced the project after activiation/change of the postprocessing step, the first time (when the problem occured), right?

bastion-0078, 2019-03-10:
I´ve to resliced it, when I change the postprocessing. Otherwise I can´t print it.

Ghostkeeper, 2019-03-11:
It seems like the USB printing plug-in didn't apply the pre-printing steps at all (such as postprocessing). I've made it call the g-code writer in order to do this automatically.

closed 2019-03-13 by nallath

Ghostkeeper, 2019-03-14:
This just got fixed in Master, so it should be working in Cura 4.1 (not in 4.0, I'm afraid).

PR #5417 (2019-03-13): Fix post-processing scripts when printing via USB
The g-code writer processes the post-processing scripts properly, so get the g-code from that plug-in.

Fixes #5410.