Ultimaker/Cura #5411: prime_tower_brim_enable error in master branches

paukstelis, 2019-03-09:
**Application Version**

Current master branch of CuraEngine and Cura
**Platform**

Linux

Everything builds properly for me, but I see the following error when trying to slice using current master branches of Cura and CuraEngine:

> DEBUG - [Thread-12] UM.Backend.Backend._backendLog [101]: [Backend] [ERROR] Trying to retrieve setting with no value given: 'prime_tower_brim_enable'

Is this expected, and if so which up-to-date branches should be used?

rburema, 2019-03-10:
Hi @paukstelis ,
Odd, master/master should work, as that feature has been merged in both repo's a while ago (but not, say, 4.0/master, as it's a prospective 4.1 feature). Also, keep in mind that `fdmprinter.def.json` has also been updated.

What's your `CuraVersion.py` say?

P.S. If you aren't already, please also use the latest version (or other appropriate branch) of Uranium!

paukstelis, 2019-03-10:
I nuked everything and restarted and it seems to work fine. I did check versions and everything seemed to be correct, so I'm not sure what was wrong. Closing this.

closed 2019-03-10 by paukstelis