Ultimaker/Cura #5412: Extruder before/after gcode missing

paukstelis, 2019-03-09:
**Application Version**

4.0 (current master branch of CuraEngine and CURA-5864_prime_tower_brim_enable which is the only one that works with CuraEngine for me)
**Platform**

Linux x64
**Printer**

Custom
**Steps to Reproduce**

Use Extruder specific gcode
**Actual Results**

Extruder specific gcode becomes garbage, with extra M82/M83 commands
**Expected results**

Extruder gcode is inserted with M82/M83 commands

Here are my custom Gcode commands for this extruder:
[image]

However, this is what I see in the gcode:

> G1 F3000 E3
> G1 F2400 X339.031 Y250.412 E0.04915
> G1 X339.101 Y250.342
> G1 F1200 E-7
> M82 ;absolute extrusion mode
> �d��U
> M83 ;relative extrusion mode
> T0
> M82 ;absolute extrusion mode
>  (�U
> M83 ;relative extrusion mode
> M105
> M109 S190
> M104 T4 S180

I believe the gibberish is where the before/after gcode segments are supposed to be. Its not at all clear why the M82/M83s are in there.

paukstelis, 2019-03-09:
I just noticed in the terminal that all of the before/after gcode is being printed out there every time it should be written:

> z_seam_y="1080" -s machine_extruder_start_code="G91; relative
> 2019-03-09 11:58:48,388 - DEBUG - [Thread-21] UM.Backend.Backend._backendLog [101]: [Backend] G0 X-5 Y-10
> 2019-03-09 11:58:48,388 - DEBUG - [Thread-21] UM.Backend.Backend._backendLog [101]: [Backend] M616 T0
> 2019-03-09 11:58:48,388 - DEBUG - [Thread-21] UM.Backend.Backend._backendLog [101]: [Backend] M109 S190
> 2019-03-09 11:58:48,389 - DEBUG - [Thread-21] UM.Backend.Backend._backendLog [101]: [Backend] G90; absolute" -s machine_nozzle_offset_x="0" -s support_interface_density="75"

rburema, 2019-03-10:
@paukstelis ,
`CURA-5864_prime_tower_brim_enable` was (merged and) deleted a while ago, have you pulled recently? The gibberish does ring a bell, it may be something that was fixed recently, but I can't find the ticket now :-/ It'll be really hard to debug these sort of problems if the versions of the frontend/uranium/engine are mismatched! I know you haven't gotten it working yet, but it would be really helpful if this where reproducable in the current masters.
P.S. That reminds me, we've changed libArcus as well in the (relatively) recent past.

paukstelis, 2019-03-10:
Went back and recloned and rebuilt master of CuraEngine, Cura, latest Uranium and libArcus as well. The issue still exists. The only thing I didn't redo was protobuf, but I'll go back and see what happens with that.

paukstelis, 2019-03-10:
This looks like a duplicate of https://github.com/Ultimaker/Cura/issues/5311

Ghostkeeper, 2019-03-11:
Indeed, closing this as duplicate. We've found this too.

closed 2019-03-11 by Ghostkeeper

Ghostkeeper, 2019-03-11:
> I just noticed in the terminal that all of the before/after gcode is being printed out there every time it should be written:

This is because there are newlines in the setting value and the front-end puts its own prefix before every line that the back-end needs to log. Nothing to worry about.