Ultimaker/Cura #5413: Cura 3.6.0 linux mint 19 Ender 3 not connecting

AT3chnicality, 2019-03-09:
**Application Version**
3.6.0

**Platform**
Linux Mint 19

**Printer**
Creality Ender-3

**Steps to Reproduce**
plug in machine, nothing occurs.

**Actual Results**

**Expected results**

**Additional Information**
I've tried the autodetectbaudjob.py change as was talked about in the windows version of the problem and the py only showed up on my machine once and now it's non existent. I tried reinstalling Cura and it still doesn't exist. I've searched the whole system as both root and non root. 

I tried going into the plugins folder --> /Home/u$ername/.local/share/cura/3.6/plugins  and it's empty.

Ghostkeeper, 2019-03-11:
A log file may help.

Which `autodetectbaudjob.py` change were you talking about? The `AutoDetectBaudJob.py` file is in the USBPrinting plug-in which gets shipped along with Cura. That means that it's packed inside your .AppImage file. The AppImage file is an ISO image so you could edit it. It will not be copied to your .local folder.

AT3chnicality, 2019-03-12:
> A log file may help.
>
> Which `autodetectbaudjob.py` change were you talking about? The `AutoDetectBaudJob.py` file is in the USBPrinting plug-in which gets shipped along with Cura. That means that [...]

Not sure which log file would be useful since it doesn't know it's not working.

and the fix i'm talking about is here --> https://github.com/Ultimaker/Cura/issues/4196 it's basically a change to it's call and response wait times for checking if the printer is connected.

I've never had to edit an iso img before. Most complex thing i've done in that department is registery editing back on windows 7.

As for specific info point me at the thing and i'll find it for you.

Ghostkeeper, 2019-03-13:
For Cura 4.1, we've applied some changes suggested in that thread: https://github.com/Ultimaker/Cura/pull/4825

Hopefully this allows you to get a connection more easily.

This is how you can find the Cura log on your system: https://github.com/Ultimaker/Cura#logging-issues

AT3chnicality, 2019-03-19:
> For Cura 4.1, we've applied some changes suggested in that thread: #4825
> 
> Hopefully this allows you to get a connection more easily.
> 
> This is how you can find the Cura log on your system: https://github.com/Ultimaker/Cura#logging-issues

**1.** OK so, still with a fresh cura, when i open the program this prints into the log. I emboldened the things that seem relevant but you know what you're looking for more than i. This is without plugging in my printer.

<details>
  <summary>Click to expand</summary>

2019-03-19 05:06:00,250 - DEBUG - [Thread-2] UM.Mesh.MeshData.calculateNormalsFromVertices [444]: Calculating normals took 0.0013852119445800781 seconds
2019-03-19 05:06:00,251 - DEBUG - [Thread-2] STLReader.STLReader._read [89]: Loaded a mesh with 25188 vertices
2019-03-19 05:06:01,013 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin._createView [279]: Creating post processing plugin view.
2019-03-19 05:06:01,305 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [172]: Loading theme file: /tmp/.mount_kGgXwr/usr/bin/resources/themes/cura-light/theme.json
2019-03-19 05:06:01,530 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [244]: Loaded theme /tmp/.mount_kGgXwr/usr/bin/resources/themes/cura-light
2019-03-19 05:06:01,621 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin._createView [289]: Post processing view created.
2019-03-19 05:06:01,944 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
_**2019-03-19 05:06:01,948 - INFO - [MainThread] cura.Machines.QualityManager.getQualityChangesGroups [179]: Cannot find node for machine def [fdmprinter] in QualityChanges lookup table**_
2019-03-19 05:06:01,952 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [50]: Updating QualityProfilesDropDownMenuModel.
2019-03-19 05:06:01,953 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container coarse
2019-03-19 05:06:01,958 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [coarse] added.
2019-03-19 05:06:01,995 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container draft
2019-03-19 05:06:02,004 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [draft] added.
2019-03-19 05:06:02,040 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container extra_coarse
2019-03-19 05:06:02,046 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [extra_coarse] added.
2019-03-19 05:06:02,084 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container fast
2019-03-19 05:06:02,091 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [fast] added.
2019-03-19 05:06:02,173 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container high
2019-03-19 05:06:02,179 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [high] added.
2019-03-19 05:06:02,219 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container normal
2019-03-19 05:06:02,224 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [normal] added.
2019-03-19 05:06:02,228 - DEBUG - [MainThread] cura.Machines.Models.BuildPlateModel._update [33]: Updating BuildPlateModel.
2019-03-19 05:06:02,532 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///tmp/.mount_kGgXwr/usr/bin/resources/qml/Topbar.qml:19: TypeError: Cannot read property 'stageId' of null
2019-03-19 05:06:02,535 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///tmp/.mount_kGgXwr/usr/bin/resources/qml/Topbar.qml:104: TypeError: Cannot read property 'stageId' of null
2019-03-19 05:06:02,538 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///tmp/.mount_kGgXwr/usr/bin/resources/qml/Topbar.qml:175: TypeError: Cannot read property 'stageId' of null
2019-03-19 05:06:02,540 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///tmp/.mount_kGgXwr/usr/bin/resources/qml/Cura.qml:444: TypeError: Cannot read property 'mainComponent' of null
2019-03-19 05:06:02,543 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///tmp/.mount_kGgXwr/usr/bin/resources/qml/Cura.qml:438: TypeError: Cannot read property 'mainComponent' of null
2019-03-19 05:06:02,546 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///tmp/.mount_kGgXwr/usr/bin/resources/qml/Cura.qml:476: TypeError: Cannot read property 'sidebarComponent' of null
2019-03-19 05:06:02,550 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///tmp/.mount_kGgXwr/usr/bin/resources/qml/Cura.qml:508: TypeError: Cannot read property 'sidebarComponent' of null
2019-03-19 05:06:02,560 - DEBUG - [MainThread] UM3NetworkPrinting.src.DiscoverUM3Action._createAdditionalComponentsView [171]: Creating additional ui components for UM3.
2019-03-19 05:06:02,647 - DEBUG - [MainThread] UM3NetworkPrinting.src.DiscoverUM3Action.startDiscovery [43]: Starting device discovery.
2019-03-19 05:06:02,660 - DEBUG - [MainThread] UM.Controller.setActiveStage [159]: Setting active stage to PrepareStage
[... cut ...]
2019-03-19 05:31:19,088 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [417]: Shutting down cura
2019-03-19 05:31:19,104 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [293]: Attempting to kill the engine process
2019-03-19 05:31:19,104 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [299]: Killing engine process
2019-03-19 05:31:19,105 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [302]: Engine process is killed. Received return code -15

</details>

fieldOfView, 2019-03-19:
The user that runs Cura does not seem to have access to the serial port.

Ghostkeeper, 2019-03-19:
> PermissionError: [Errno 13] Permission denied: '/dev/ttyUSB0'

This is the root of the problem, I'm guessing. Maybe another application is holding the handle to this port? Cura itself does that wrongly, so maybe you have multiple instances of Cura running?

closed 2020-06-20 by no-response[bot]