Ultimaker/Cura #5414: 3.6.0 Is there any way to use .amf files?

FlailAway, 2019-03-09:
Thanks for Cura!

**Application Version**
3.6.0

**Platform**
Mint 19.1/Cinnamon

**Printer**
Any

**Steps to Reproduce**
None needed

**Actual Results**
I cannot see the file

**Expected results**
I can see and use the file

**Additional Information**
Is there any way to use .amf files to slice? Slic3r-PE seems to be happy with them and it would save me some duplication as I often have to save files in both formats. It is handy to be able to use Slic3r-PE to repair some outside (Thingi etc) files and then export them back out to the same .amf format and then open them in Cura as I get much better prints with Cura than Slic3r.

Thanks

fieldOfView, 2019-03-09:
No. Not unless someone writes a plugin for it.

FlailAway, 2019-03-09:
Thanks, how hard would that be? Could I use the Slic3r-PE code as a starting point? I am a programmer but not in this field. Where will I find the Cura Plug in requirements?

fieldOfView, 2019-03-10:
It would be 5 hard.  Kidding aside, without knowing your skill level, it is hard to say how hard it will be.

Slic3r and Cura are so different that I don't think you'll gain much from staring at the Slic3r code. I think the X3D reader plugin will be a better starting point. Like AMF, X3D is XML-based.
https://github.com/Ultimaker/Cura/tree/master/plugins/X3DReader
For decompressing zipped AMF files (zipping them is optional), you may want to have a look at the gzipped gcode reader (though you may have to do some detection-magic, since the AMF format does not have a different file extension for zipped and non-zipped files).
https://github.com/Ultimaker/Cura/tree/master/plugins/GCodeGzReader

General information about creating plugins can be found in the Uranium wiki:
https://github.com/Ultimaker/Uranium/wiki/Creating-plugins

FlailAway, 2019-03-10:
Thanks for the detailed reply. My skill level is very high in machine control systems etc, but web-stuff ain't in there anywhere, so XML largely looks like spaghetti to me. Too old and too little spare time to learn another language to the depth I am guessing it would be required. I will suffer in silence then. :)

Ghostkeeper, 2019-03-11:
In Python we don't deal with the XML format itself. We just call something like `doc = xml.etree.ElementTree.parse("data.xml")` and then we get a tree data structure in the `doc` variable.

Ghostkeeper, 2019-03-11:
We intended to add AMF support at some point. But that requires buying the spec, since they don't give it out for free even though it's an open specification. And we never got around to do that.

FlailAway, 2019-03-12:
> We intended to add AMF support at some point. But that requires buying the spec, since they don't give it out for free.

How much would that cost? My deeply hidden altruistic streak may be overriden long enough to make a donation.

Ghostkeeper, 2019-03-13:
60 euros iirc. It's absolutely nothing that imposes a barrier for Ultimaker, so don't bother.

But then of course we'd also have to implement it, and finding time for that is difficult. I considered buying the spec myself as well and implementing it from there but also couldn't find the time for it yet. Maybe once the SVG reader plug-in I'm working on is a bit more fully featured, I'll give it another go.

I couldn't find any pre-made library for Python that loads AMF. Even Trimesh seems to be unable to do that.

fieldOfView, 2019-03-13:
Do you even need the spec though? It is a fairly simple format. Here's how legacy Cura read it:
https://github.com/daid/LegacyCura/blob/SteamEngine/Cura/util/meshLoaders/amf.py

smartavionics, 2019-03-13:
And openscad also has an AMF reader (in C++ but it's another source).

nallath, 2019-03-13:
It's a partial implementation of AMF where Daid interpreted the situations where the spec was contradicting itself. He should have the spec lying around somewhere.

FlailAway, 2019-03-13:
> Do you even need the spec though? It is a fairly simple format. Here's how legacy Cura read it:
> https://github.com/daid/LegacyCura/blob/SteamEngine/Cura/util/meshLoaders/amf.py

Is there any quick and dirty way that this could be adapted as a Plug-in for 3.6.0? Even the partial Implementation would probably be good enough for what I need.

fieldOfView, 2019-03-14:
I know that I could lift the legacy code and put it in a Cura plugin in a couple of hours. But I already have 10 plugins to maintain (as a community contributor), and frankly you are the first one to ask for AMF file compatibility in a loong time.

fieldOfView, 2019-03-14:
I decided to put my money where my mouth is, and will take a stab at making an AMF Reader plugin. Some links to .amf files for testing are appreciated.

Ghostkeeper, 2019-03-14:
I would be interested in putting that plug-in in the main Cura code base if you're successful. That would take the burden of maintenance off your shoulders. And compatibility with different file formats is not something that gets in the way so it doesn't need to be optional.

nallath, 2019-03-14:
The file loaders haven't changed much the last time, so if you do decide to put it in the mainline, it's hardly any work for us to keep it up to date (I expect that most updates will be "bump version number")

fieldOfView, 2019-03-14:
See #5429

PR #5429 (2019-04-30): Add AMF reader plugin
This PR adds support for reading AMF files. It is based on the implementation of legacy Cura, and is similarly incomplete.

https://github.com/daid/LegacyCura/blob/ad7641e059048c7dcb25da1f47c0a7e95e7f4f7c/Cura/util/meshLoaders/amf.py

Multiple meshes inside a single file are grouped together so they are not pushed apart.

Fixes #5414
closed 2019-04-30 by nallath