Ultimaker/Cura #5415: [3.6.0] Different slicing results with Mac and Windows

Levious, 2019-03-10:
**Application Version**
3.6.0

**Platform**
Win7 SP1 and MacOS Mojave 10.14

**Printer**
Custom ULTI

**Steps to Reproduce**
Open project, slice than compare g-code or just print

**Actual Results**
under-extrusion on model sliced with Win Cura, despite the fact that according to the g code the printer squeezes out more plastic

**Expected results**
identical printing results and g code

**Additional Information**
Today I will print these 2 g code and attach a photo
[Mac win Cura.zip](https://github.com/Ultimaker/Cura/files/2949649/Mac.win.Cura.zip)

Ghostkeeper, 2019-03-11:
Even the start g-code is different. I think that your Windows computer already had a Custom FFF Printer added and you didn't let it override that properly, because the configuration that I see in your g-code is not the same as the one in the 3MF file.

Levious, 2019-03-12:
> Even the start g-code is different. I think that your Windows computer already had a Custom FFF Printer added and you didn't let it override that properly, because the configuration that I see in your g-code is not the same as the one in the 3MF file.

Thanks, got it. And how should I fix this? Where should I look?

Ghostkeeper, 2019-03-13:
In the preferences, make sure that "Show summary dialog when importing project" is ticked. Then import the 3MF file in Windows. You need to import it as a project, and make sure to "import as new" printer, so that a new printer gets added.

For the start g-code specifically, that'll be located under Manage Printers... then Machine Settings. The rest of the settings are just in the sidebar (<=3.6) or the settings panel (>=4.0).

Ghostkeeper, 2019-09-17:
We'll assume that was the issue.

closed 2019-09-17 by Ghostkeeper