Ultimaker/Cura #5416: [3.6.0] Fail to connect

sk8nfool, 2019-03-10:
**Application Version**
Cura 3.6.0

**Platform**
Ubuntu 18.04

**Printer**
Creality Ender 2

**Steps to Reproduce**
Turn off printer
Start Cura (note that printer is selected but not connected)
Load a model (may not be required)
Turn on printer

**Actual Results**
Printer remains selected but not connected.

**Expected results**
Either

- Printer discovered and connected.
- The ability to force connection or discovery

**Additional Information**
Here's a screen shot (after a couple of minutes of delay). Notice that the printer is selected (show in upper right corner) but not connected (stated twice on right side). Checking with "lsof -c cura" shows that Cura has the serial device open (BUT NOT EXCLUSIVELY! see bug #5207 and 5398).
[image]

Ghostkeeper, 2019-03-11:
We can't debug this without having a physical Ender 2 printer. I hope someone can find out what the problem is here and solve it for us.

nallath, 2019-03-14:
Also; if someone does want to solve it, that person will need log files.

sk8nfool, 2019-06-15:
actually I don't believe this is limited to the Ender-2.

wrightwells, 2019-06-17:
Same issue on Ender 3.

mschilt, 2022-02-06:
Works great for me with Ender 3 on Cura 4.13 on Ubuntu 21. I can turn on/off the printer all the times and it gets instantly detected.
Are you sure this issue persists in the latest version? Can we close this issue?

closed 2022-02-19 by Ghostkeeper