Ultimaker/Cura #5420: /usr/share/cura-lulzbot/cura-lulzbot: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /usr/share/cura-lulzbot/cura-lulzbot)

MrDadaGuy, 2019-03-11:
**Application Version**

apt-cache policy cura-lulzbot
cura-lulzbot:
  Installed: 3.6.3
  Candidate: 3.6.3
  Version table:
 *** 3.6.3 500
        500 http://download.alephobjects.com/ao/aodeb buster/main amd64 Packages
        100 /var/lib/dpkg/status
     3.2.32 500
        500 http://download.alephobjects.com/ao/aodeb buster/main amd64 Packages
     3.2.27 500
        500 http://download.alephobjects.com/ao/aodeb buster/main amd64 Packages
     3.2.23 500
        500 http://download.alephobjects.com/ao/aodeb buster/main amd64 Packages
     3.2.21 500
        500 http://download.alephobjects.com/ao/aodeb buster/main amd64 Packages

**Platform**

Ubuntu 18.04

**Printer**

Lulzbot Taz6 - but the app doesn't stay open long enough for this to matter

**Steps to Reproduce**

run "cura-lulzbot" from terminal

**Actual Results**

$ cura-lulzbot 
/usr/share/cura-lulzbot/cura-lulzbot: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /usr/share/cura-lulzbot/cura-lulzbot)

**Expected results**

ideally the app opens and i can 3d print some cool stuff

**Additional Information**

MrDadaGuy, 2019-03-11:
I'll answer my own question -- aleph repo doesn't have a "bionic" so I guess I had used "buster"

a new release of cura-lulzbot to 3.6.3 must be depending on version of glibc that I don't yet have on my machine (because it's bionic).  

I rolled back to previous version and all's well:

`sudo apt install cura-lulzbot=3.2.32
`

closed 2019-03-11 by MrDadaGuy

nallath, 2019-03-11:
Cura Lulzbot is also not maintained by us (it's a fork from this version of Cura)