Ultimaker/Cura #5421: Create a pip package

KOLANICH, 2019-03-11:
It would be nice to use cura from withkn existing python installation. To do it we need a `setup.py` and `setup.cfg`, providing the instructions to `setuptools` on how to build a wheel.

Ghostkeeper, 2019-03-13:
That would be kind of involved, since you'd also need CuraEngine which needs to be compiled with MinGW specifically, while all the Python modules need to be compiled with MSVC. Also, this package would need to contain libArcus, Geos, libXml, lxml, OpenBlas, OpenSSL, Protobuf, PyQt and Qt, libSavitar, SIP, XZ, Uranium, cura-binary-data, fdm_materials, libCharon and Cura itself.

The Cura build scripts ([here](https://github.com/Ultimaker/cura-build) and [here](https://github.com/Ultimaker/cura-build-environment)) are several thousand lines of code right now, though admittedly this also contains packaging and signing code which would be unnecessary for Pip.

I'm also wondering what the Cura package would do once installed in Pip. Would there be some sort of "run()" function that starts Cura?

KOLANICH, 2019-03-13:
>since you'd also need CuraEngine which needs to be compiled with MinGW specifically, while all the Python modules need to be compiled with MSVC

XGBoost built with MinGW-w64 works quite fine with python built with MSVC. We need to try first.

>Would the __main__ function just start Cura? Is that all?

No. __main__.py is needed importing and calling your `__main__` (or any function with other name).
This would make `pythin -m cura` work.

In order to make just `cura` work, an entry_point `console_scripts` is needed pointing to that function, like in https://github.com/KOLANICH/HDDModelDecoder.py/blob/master/setup.cfg#L31