Ultimaker/Cura #5422: Cura Connect: 3MF from another computer cannot print over network

BenoitValin, 2019-03-12:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2019-03-13:
Did you expect this:

1. Connect to printer on Computer A.
2. Save 3MF.
3. Load 3MF on Computer B.
4. Print via network to the printer that you connected to on Computer A.

Or did you expect this?

1. Connect to printer on Computer **B**.
2. Save 3MF from Computer A.
3. Load 3MF on Computer B.
4. Print via network to the printer that you were already connected to on Computer B.

We could implement one or the other, but not both. Currently (without testing) I'd expect the first scenario to work.

BenoitValin, 2019-03-13:
Hi, 

  Both scenarios do not work, but the second scenario is what i am referring to
1 - Save 3MF from Computer A (prepare a build for a client)
2- Connect to printer on Computer B (client opens cura and connect to their Cura Connect Host)
3- Load 3MF on Computer B (client opens 3MF I sent them)
4 - Print via network to the printer that you were already connected to on Computer B (client prints on a printer connected to their Cura Connect Host)

  The problem is that at step 3, Cura creates a new local printer (even when no local printers are defined). Switching from a local printer to a Cura Connect Host deletes all the parameter over-rides specified in the 3MF to that of the last used on that Cura Connected printer. 

  The only way is saving to USB and printing with USB.

  This applies to 2 computer on the same LAN connected to the same Cura Connect Host, or computers in different locations (i.e. Ultimaker office and client's office)

I hope this clarifies the matter.

nallath, 2020-11-26:
This has been fixed in the meantime.

closed 2020-11-26 by nallath