Ultimaker/Cura #5423: Gap Filling generations needs optimization

NicholasTracy, 2019-03-12:
**Application Version**

3.6.0
**Platform**

Windows 10
**Printer**

Tevo Tarantula
**Steps to Reproduce**

Created a model that has a wall section that is 1.2mm thick. 
Nozzle diameter is 0.4mm
Line Width is 0,4mm
Fill gaps setting set to Everywhere
Slice.

**Actual Results**

Tiny zig zag patterns are used to fill the gaps.
Print Time is greatly increased
[image]

**Expected results**

The Gaps should be instead filled with a straight wall section.
Print time should be much less.
**Additional Information**

Turning off filling gaps leaves a void the width of a singe wall line.
[image]

smartavionics, 2019-03-12:
This has received some attention in 4.0 so please try the beta and see if it is better.

Alternatively, you can try one of my releases (also based on 4.0) that has a different implementation of wall gap filling and thin wall support along with other useful features and bug fixes. You can find releases for Linux and Windows at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

NicholasTracy, 2019-03-12:
Thanks I will take a look at it. I assume the most recent .exe file is safe to use?

smartavionics, 2019-03-12:
Hi, if you are referring to my releases, then all I can say is that I try not to release anything that has any known bad problems, but, of course, I can't exhaustively test and the releases are based on the Ultimaker master branches so it's a case of YMMV. If you find a problem that is specific to my release (compared to the Ultimaker release), please open an issue on my Cura repo (smartavionics/Cura).

NicholasTracy, 2019-03-12:
Will do. I will give it a try. I will also try 4.0 beta to see if its any better there.

NicholasTracy, 2019-03-18:
The final verdict on this:
4.0 seems to be better at gap filling but some areas still cause issues. particularly when there is a gap wider than 2 wall thicknesses, it still does micro zigzags.
I have had the best luck so far with @smartavionics release all around.

closed 2019-03-18 by NicholasTracy