Ultimaker/Cura #5424: How to over-extrude my infills, only my infills

swatti87, 2019-03-12:
Is there a separate setting for infill extrusion?

I do 100% infilled parts. My walls are 0.2 tall but my infilling is 0.6 to save some time. I want to over-extude a bit so the lines stick better to the layer below.

I'm running this rig very hot and very fast on purpose. Infill "Density" does not help since it makes the lines just closer to each other and infill "thickness" i need to keep where it is, but i need to extrude a biiiit more to "squish" the infill together with the previous layer. Kinda like over-extruding a first layer.

smartavionics, 2019-03-12:
You don't say what your printer is or how fast you are printing but I would guess that you are seeing underextrusion due to the non-linearity of your extruder+hotend. What I mean by that is that if you calibrate your extruder steps to get, say, 100mm of filament extruded at 1mm/S feedrate then at, say 5mm/S feedrate you will actually achieve a lot less than 100mm of filament extruded. This non-linearity effect is worse for small (< 0.4) nozzles. I know that RepRap firmware supports non-linear extrusion, perhaps other printers (and yours) can also do that?

Back to your question, there's no way in Cura to specify an infill flow rate modifier.

swatti87, 2019-03-12:
Custom rig. Tevo BW chassis. "big cartesian" 

160mm/s 
2500acc
0.6mm nozzle
Copper Volcano
Bondtech extruder
SCV @ 9mm/s
PETG @ 260c
Outer walls 0.2mm layers
100% Infill at 0.6mm layers
Current speed is _mere_ 160mm/s at 2500acc. 

I tried making the lines wider, overlap a bit more via 110% "density" but it doesnt really achieve the over-extrusion i want. This makes them stick more to the line on their side, not the lower layer.
I'd be veeeeery happy to do like ~90% infill with massively over-extruded infills so the first part they stick to is the previous layer. 

Spaghetti infill has "extra volume" for it. Its pretty much an over-extrusion percentage... I want that to my regular infill :(
(spaghetti infill is buggy and poops out in large parts and when laid between multiple wall-layers)

smartavionics, 2019-03-12:
>SCV @ 9mm/s

Sorry, what does that mean?

Ghostkeeper, 2019-03-13:
We have some plans to adjust flow for infill separately: https://github.com/Ultimaker/CuraEngine/pull/874

Ghostkeeper, 2019-03-13:
For now I'd suggest increasing the line width for infill and then increasing the infill density by the same ratio.

swatti87, 2019-03-13:
SCV : Square Corner Velocity 
Klipper FW space magic.

> We have some plans to adjust flow for infill separately: [Ultimaker/CuraEngine#874](https://github.com/Ultimaker/CuraEngine/pull/874)

Out when? Soooo awasum! 
I hope there will also be a build that lets you adjust all setting set visible. For the more advanced users.

> For now I'd suggest increasing the line width for infill and then increasing the infill density by the same ratio.

Tried, only minor help. Ends up creating other minor issues.

Ghostkeeper, 2019-03-14:
> Tried, only minor help. Ends up creating other minor issues.

Then I'm afraid that increasing flow will have the same issues. They are logically equivalent.

swatti87, 2019-03-14:
Line width for infill has handfull of small differences. It mainly makes the lines stick the sides of the previous ones. It doesnt make em much higher in Z

It also has another "side-effect" of leaving gaps where it thinks it can fit. On complex parts it leaves LOT of small gaps and there isnt an option to infill them with anything :\

swatti87, 2019-04-01:
IDEA / QUESTION:

"Two extruders" from a one
Can i add a second extruder, that actually is the first and only extruder? Kinda like a messed up later-ego :D

That extruder should have different settings for flow and if it assigned as the infill-extruder, it would over-extrude on all infills.... 

Right?

Ghostkeeper, 2019-04-02:
Yeah. If your printer actually has only one extruder, I'd expect it to skip over all the T1 commands that would otherwise switch tools. And otherwise, you can use a post-processing script to remove them.

Ghostkeeper, 2020-12-18:
The Infill Flow setting has since been implemented.

closed 2020-12-18 by Ghostkeeper