Ultimaker/Cura #5426: [Feature Request] Orthographic View

krz4tw, 2019-03-13:
Hey,
I just found [#3159](https://github.com/Ultimaker/Cura/issues/3159) and would appreciate that feature very much. I´d suggest a little mod to the old request to make it more easily implementable

I´d like to have orthographic view to be able to postion supportblockers precisely for example.

The zooming and panning should be quite easy to implement if you only enable orthographic in Top down, left, right,.... mode when you choose them via the little cube icons in the top right corner. Those 3-side projections are by far the most useful way for orthographic projection. I don´t need to orbit around in orthographic view.
Zooming could be done by "scaling" what is on the screen and panning just by doing xy offsets of whats already on screen. Once somebody uses right mouse button to orbit switch back to perspective view.
In fact, I´d make orthographic the default when you click on the cube icons.

Thanks

Thisismydigitalself, 2019-03-14:
https://github.com/Ultimaker/Cura/issues/4806

krz4tw, 2019-03-14:
I wish I would have made a notice of everytime I pushed the blender-ortho-view hotkey just out of pure habit in Cura so that I could give you a long list of situations where it would be usefull, sadly I didn´t.
I feel like this would be one of those changes that you never know you needed but don´t want to live without once you have it.

Ghostkeeper, 2019-03-14:
It's one of those things that comes back once in a while because CAD users are used to it. CAD integration is pretty high up on the vision that Ultimaker has for Cura so I expect that it won't take much longer before we get to implement this properly.

Ghostkeeper, 2019-03-20:
This is a duplicate of #2545.

closed 2019-03-20 by Ghostkeeper