Ultimaker/Cura #5427: [Feature Request] Bottom and Rear View

krz4tw, 2019-03-13:
Hey,
I just noticed that there is no way to go into a bottom or rear view via the cube icons in the top right.
There´s an extra icon for left and right view though.

My suggestion would be  to get rid of either the left or right icons and make all icons (for example) shift-clickable.
If you normally click on the front view icon, it will switch to front view. If you shift+click on front view it will go to rear view. Same for Top and left/right icons.

Thanks

Ghostkeeper, 2019-03-14:
Shouldn't be too hard to implement, but I'm concerned that it would be hard to discover. Who could guess that holding shift while clicking these buttons causes it to have a different effect?

Viewing the bottom is not a very common use case (yet). The back is a bit more common on Delta printers which don't really have a front side.

krz4tw, 2019-03-14:
You could always just add another cube icon for the rear and bottom view, the idea with a shift click was just to reduce the amount of icons up there. As long as you can get the views its fine with me.