Ultimaker/Cura #5431: [4.0.0-BETA-2] Adaptive Layers - maximum variation not honored for maximum

rwese, 2019-03-15:
**Application Version**

4.0.0-BETA-2

**Platform**

Windows 10

**Steps to Reproduce**

* Set Layer Height 0.2
* Activate "Adaptive Layers"
* Adaptive layers max. variation 0.08
* Adaptive layers variation step size 0.04

**Actual Results**
* Layer Height, 0.2 max variation 0.08, steps 0.04, Result: 0.12 - 0.30
[image]
* Max variation 0.02 result in 0.18 - 0.30
[image]

**Expected results**

* Layer Height, 0.2 Max variation 0.08 should result in 0.12 - 0.28
* Layer Height, 0.2 Max variation 0.02 should result in 0.18 - 0.22

**Additional Information**

[PI3_xyzCalibration_cube.3mf.zip](https://github.com/Ultimaker/Cura/files/2971410/PI3_xyzCalibration_cube.3mf.zip)

Did work properly on 3.6.0

Hope this is complete "enough", i will try my luck looking at the source.

rwese, 2019-03-15:
I am very very sorry, this is a user error... the initial layerheight was reset to 0.3 that threw me off.

closed 2019-03-15 by rwese

ChrisTerBeke, 2019-03-15:
Indeed the initial layer height is left untouched by this feature as its needed to ensure proper adhesion. Thanks for filing this quite detailed report though!