Ultimaker/Cura #5432: Small Bug 'Nozzle Switch Retraction Distance'

fntsmn, 2019-03-15:
**Application Version**
Cura 4.0 latested release

**Platform**
Windows 10

**Printer**
Custom, I'm making a profile for a Craftbot 3 Idex printer

**Steps to Reproduce**
I deselected the retraction in the right extruder, but it was keep using the Nozzle Switch Retraction Distance of 16mm

**Actual Results**
Was keeping the Nozzle Switch Retraction Distance even if the Retraction was not selected

**Expected results**
It should not use the Nozzle Switch Retraction Distance if the Retraction box is not selected

**Additional Information**

[image]

I cannot show the parameter when the retraction is not selected but I did checked the Gcode and was adding a 16mm retraction that wasn't suppose to be there

Ghostkeeper, 2019-03-19:
Yeah, I see in the code that it's not checking for `retraction_enabled` in CuraEngine, but it is removing the setting in the front-end. There is a discrepancy there.

Ghostkeeper, 2019-03-22:
Working now. Here's a screenshot that shows one extruder having retractions disabled and the other having it enabled. The line towards the extruder switching bay is darker when retractions are disabled now.

[image]

closed 2019-03-22 by Ghostkeeper