Ultimaker/Cura #5433: Dual Extrusion Wipe off the nozzle

lwbsbde, 2019-03-15:
CURA 3.6.0

I have a dual extruder with two nozzle.
after the first change to nozzle 2 and second change to nozzle 1, the nozzle is  wiped off the model and not on the tower.

This only happens with the first two changes of nozzles. Afterwards ,he wipes the nozzle  correctly at the tower.

Ghostkeeper, 2019-03-19:
I can't reproduce this bug. Maybe you can provide a project file that shows the issue?

lwbsbde, 2019-03-24:
[FlexiRex.zip](https://github.com/Ultimaker/Cura/files/3000848/FlexiRex.zip)

Ghostkeeper, 2020-06-30:
Okay, that was a bit unfair, mr. Bot.

Ghostkeeper, 2020-06-30:
The provided project file has the prime tower disabled though. So obviously it's not going to be wiping off the prime tower.
When I enable the prime tower, it seems to wipe properly. Even the second nozzle change. The first nozzle change is on the first layer where there is a skirt to wipe it off instead, so there it's not expected to need a prime tower.

Maybe this has been fixed in the meanwhile?

closed 2020-07-14 by no-response[bot]