Ultimaker/Cura #5434: [3.1.0] Cura reloads previous files

webware-nach-mass, 2019-03-16:
**Application Version**
3.1.0

**Platform**
Kubuntu Linux

**Printer**
N/A

**Steps to Reproduce**
Start Cura
Click "Open" button
Select .obj file -> model get loaded
Click File / New Project or Ctrl+N, accept dialog
Click "Open" button
Deselect previous .obj file, select another .obj File -> **both** selected file and previous file get loaded
Click File / New Project or Ctrl+N, accept dialog
Click "Open" button
Deselect all, select only one .obj File -> **all three** .obj files get loaded

**Actual Results**
.obj files from previous projects get loaded into new project

**Expected results**
Cura should only load the selected file, not the old ones

**Additional Information**
N/A

fieldOfView, 2019-03-16:
Have you tried 3.6? Older versions will not be fixed.

Ghostkeeper, 2019-03-19:
I remember this one. It's been fixed quite a long time ago.

closed 2019-03-19 by Ghostkeeper

webware-nach-mass, 2019-03-26:
@fieldOfView How can I install 3.6 under Kubuntu Linux?
@Ghostkeeper 3.1.0 is the version that the Kubuntu app store offers. Do you know where and how I can file a request so they go to a better version?

fieldOfView, 2019-03-26:
Use the AppImage you can download from software.ultimaker.com