Ultimaker/Cura #5435: [3.1.0] File > New Project doesn't reset display

webware-nach-mass, 2019-03-16:
**Application Version**
3.1.0

**Platform**
Kubuntu Linux

**Printer**
N/A

**Steps to Reproduce**
Load Model
Click "Prepare"
Save .gcode file
Use File > New Project or Ctrl+N

**Actual Results**
Printing time, filament length and weight remain displayed
"File saved" message remains displayed

**Expected results**
Informations relating to previous project (printing time, filament length and weight, messages) should be removed from display.

**Additional Information**
N/A

nallath, 2019-03-18:
Does this also occur with the latest versions (eg 3.6 or 4.0)?

Ghostkeeper, 2019-03-18:
Can't reproduce this in 4.0 right now.

webware-nach-mass, 2019-03-26:
@nallath How can I install Cura 3.6 or 4.0 under Kubuntu Linux?
@Ghostkeeper 4.0 on which Platform?

nallath, 2019-03-26:
You download the app image from the website, you change it's permissions so you can execute it and then run it.

Ghostkeeper, 2019-03-28:
> @Ghostkeeper 4.0 on which Platform?

This is on Ubuntu 18.10. However as this is all in Python I would not expect the behaviour to be different between my system and yours.

closed 2020-06-20 by no-response[bot]