Ultimaker/Cura #5437: Over-ride infill with thin walls?

FlailAway, 2019-03-16:
**Application Version**
3..6.0

**Platform**
Mint 19.2. Cinnamon

**Printer**
Any, I am guessing

**Steps to Reproduce**
Nozzle 0.4mm, walls in .stl at 2mm and wall-count set at 2 in Cura.

**Actual Results**
Cura will run the two walls each side, then do an infill-rattle for the single 0.4mm infill area.

**Expected results**
See that the infill is the same as a wall width and just run it.

**Additional Information**
Here I sit for a 2-hour print that Slic3r used to do in 27-minutes with 2+2+1-walls. I totally accept this is my fault as I should have changed the wall count. But, I assumed that Cura would be smart enough to realize an extra wall was preferable to an infill.

FlailAway, 2019-03-16:
UPDATE: Just tried everything I can thInk of and in Layer-View, it always prints infill in thin walls. Tried using some of the "standard" profiles, "Draft," "Fast" etc, still the same rattle along the walls. Even 0% infill gets infill in the walls.

Do I need to turn something else off?

smartavionics, 2019-03-16:
Please save the project file (File -> Save), zip the .3mf file and attach to this issue. Thanks.

FlailAway, 2019-03-17:
> Please save the project file (File -> Save), zip the .3mf file and attach to this issue. Thanks.

There is no need, just use the test file and check the Layer View. My workshop is about a quarter mile from the house and no Internet down there. I am in the house and just downloaded the latest 3.6.0 appimage and used Anycubibic Mega i3 as a "machine," at random. 

I sliced that test file using the default setting for "Basic" or whatever it is. Same issue, 1hr:15m and lots of hammering along the walls with infill. I then tried setting "show all" and tried changed things but nothing makes a difference. 

My actual print file has a bunch of those 2mm walls. Let me know if you absolutely need that 3mf file and I will walk down and get it. But, it is doing the exact same thing with this fresh install.

FlailAway, 2019-03-17:
Just playing with that test file and reducing the wall thickness in CAD to 1.6mm (an even 4-walls for Cuar) reduces the time greatly. Seems like it is only when there is an odd multiple of walls it insists on using the infill instead of replacing it with another wall.

smartavionics, 2019-03-17:
I only wanted the file because it would have contained the settings you used to slice. Without the settings, it's very hard to know exactly what is going on. BTW, I recommend that you try the 4.0 beta as it has some changes that will affect the filling of gaps between walls which is what you are doing.

FlailAway, 2019-03-17:
OK, but the point I was trying to make is that had you sliced the test file with whatever settings you have, you too would get the same infill issue. But here's the files.

[wall01a.obj.zip](https://github.com/Ultimaker/Cura/files/2975752/wall01a.obj.zip)

Is there an appimage of the beta? Where?

smartavionics, 2019-03-17:
Thanks for the files. The Ultimaker beta 4.0 release is available from their website...

https://ultimaker.com/en/products/ultimaker-cura-software/list

You may also care to try one of my releases which has an alternative implementation of the wall gap filling and thin wall printing (along with a host of other bug fixes and tweaks). My releases can be installed alongside the Ultimaker releases without conflicting. They can be found at...

https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

Ghostkeeper, 2019-03-18:
Duplicate of #5445. It's not actually infill here, but small gap filling where an additional line doesn't fit.

closed 2019-03-18 by Ghostkeeper