Ultimaker/Cura #5440: Suggestion for supports in Cura

pauloreal1, 2019-03-17:
This is a suggestion, when slicing an object without supports and there is something that prints in mid-air, CURA should issue a warning stating the layer that support was needed because there is no connection from base object because of the height it is being printed (not touching base)
[image]
 

**Application Version**

3.6.0

Ghostkeeper, 2019-03-19:
We colour it bright red. That should serve as a warning, I'd say.

pauloreal1, 2019-03-19:
A lot bright red areas were shown in this design they weren't needed. 
I'm just making a suggestion about the hanging in the air printing not the others that are attached.
Thank you

pauloreal1, 2019-03-19:
[image]

Ghostkeeper, 2019-03-19:
We currently do have some false positives there for very thin/short overhang areas (like underneath the text in the cat's coin). Those usually print fine without support. But to find those areas we basically have to slice completely, so we can't show that in the prepare mode then.