Ultimaker/Cura #5441: [4.0.0-BETA2] Loading of file not abortable

thopiekar, 2019-03-17:
**Application Version**
4.0.0-BETA2

**Platform**
Doesn't matter

**Printer**
Doesn't matter as well 

**Steps to Reproduce**
Load a large file
(Realize it is not what you want to load)
Close the progressbar of the load process in hope it will abort the load of the file

**Actual Results**
The file is still being loaded

**Expected results**
By closing the load message, the load process should also be stopped.

**Additional Information**
Not really..

Ghostkeeper, 2019-03-18:
Yeah, this reproduces for me as well.

diegopradogesto, 2019-03-22:
For internal reference: CURA-6405

diegopradogesto, 2019-03-22:
I'll set it as a new feature since there is no feature in Cura to kill a job (thread) when it's already running.

thopiekar, 2019-03-22:
@diegopradogesto Didn't know that. Always had the feeling that is it there because you can close the message about the loaded file.
One option could be to hide the cross as long as the feature is not there. Might be something for 4.0.1?

diegopradogesto, 2019-03-26:
Well, the cross is there because is intended to close the message, not to stop the operation (this is what I would expect).

We also don't plan to release 4.0.1, so... it's in the backlog, but I don't see urgency on implementing this since that was the case for a long time now. 

We will keep you up to date when we finally add this. You are also free to contribute 😜

thopiekar, 2019-03-26:
Well, I actually expected always the different, but never really noticed it 😛 

Fine for me. Also wouldn't say that this is anything major that needs a fix since nobody ever complained about it really.

Haha, thanks. To be honest I have no idea how to really kill a thread. All I know is that it might be possible to add a boolean which tells whether to abort the process. Doing a kill like `kill -9` would bring the CAD plugins which are using Windows Automation into an undefined condition, I think. 🤔

Ghostkeeper, 2019-03-28:
You can't use the `kill` command because it's a thread, not a separate process.

Ghostkeeper, 2019-06-03:
Our project manager removed this issue from our planning. We won't be able to spend time on fixing this any time soon.

I'm proposing the idea to remove the X button instead, then.

diegopradogesto, 2019-06-24:
Sorry @thopiekar , too many things to work on, and this is not an easy one. Open to ideas, though.

closed 2023-07-13 by github-actions[bot]