Ultimaker/Cura #5443: Cura 4.0 stuck window

captNtinker, 2019-03-18:
[image]
Project screen loads off my screen, it's anchored to the Right side and will not move ... I can't drag it and when maximized does not let me select the menu items (they son't pull down) program is useless in this manner :(

fieldOfView, 2019-03-18:
Your screenshot did not get through. Could you try again? Note that you have to visit this issue to add images (https://github.com/Ultimaker/Cura/issues/5443), attaching them through email won't work.

fieldOfView, 2019-03-18:
I guess you meant this image:
[image]

captNtinker, 2019-03-18:
Yes, that’s the image!

fieldOfView, 2019-03-18:
Please continue the discussion here, not in the other issue.

I am afraid your issue is a sideeffect of your screen-resolution; since the redesign the minimum vertical screensize supported seems to be 800 pixels (eg 1280 x 800).

Ghostkeeper, 2019-03-18:
Yeah. I don't think we'll reduce the minimum screen size any further than what it is in the 2nd beta.

There is a workaround though. First close Cura. Then open the file `C:\Users\<username>\AppData\Roaming\cura\4.0\cura.cfg` with a text editor and change the following preferences:

```
window_left = 0
window_top = 0
```

Setting them both to 0 should cause the window to be aligned to the top left of your screen, causing more of the Cura window to be visible.

closed 2026-09-07 by GregValiant