Ultimaker/Cura #5444: Cura 3.6.0 Retract at Layer Change setting is ignored

riodoro1, 2019-03-18:
**Application Version**
Cura 3.6.0

**Platform**
macOS Mojave 10.14.3

**Printer**
Velleman Vertex Delta

**Steps to Reproduce**
Search for the "Retract at Layer Change" setting and uncheck the checkbox. Slice the model and save the file.

**Actual Results**
The resulting GCode contains retraction commands before starting every layer.

**Expected results**
There should not be layer change retractions with this setting unchecked.

**Additional Information**
Slicing the same model with only "Retract at Layer Change" setting different produces very similar Gcode with minor differences. Disabling retraction altogether does remove all retraction moves including at the layer change.

smartavionics, 2019-03-18:
Do you have the retract before outer wall setting enabled?

Why don't you save the project file, zip it up and attach to this issue so we can check your settings. Thanks.

riodoro1, 2019-03-18:
[Retract_at_layer_change_disabled.3mf.zip](https://github.com/Ultimaker/Cura/files/2977192/Retract_at_layer_change_disabled.3mf.zip)
Here you go. Seems to me "Retract befor outer wall" is disabled.

smartavionics, 2019-03-18:
Thanks for the project file, I can confirm the bug. Will investigate...

riodoro1, 2019-03-18:
Great, thanks for very fast response 👍

smartavionics, 2019-03-18:
I have submitted a fix but don't expect it to appear in Cura any time soon.

closed 2019-03-18 by Ghostkeeper