Ultimaker/Cura #5445: 4.0.0 BETA 2: Thin Walls suddenly broken

BagelOrb, 2019-03-18:
**Application Version**
Cura 4.0.0 Beta 2

**Platform**
Linux

**Printer**
UM3

**Steps to Reproduce**
Enable Print Thin Walls
Load model
[image]
[UM3_heman hueman 3__zw.3mf.rename.zip](https://github.com/Ultimaker/Cura/files/2977354/UM3_heman.hueman.3__zw.3mf.rename.zip)

**Actual Results**
Thin walls are only jagged lines

**Expected results**
Jagged lines are combined into smooth lines

**Additional Information**
I have tested and verified that this was not a problem in Cura 3.4, Cura 3.6, or even Cura 4.0.0 BETA 1

smartavionics, 2019-03-18:
Hi Tim, the thin wall (and wall gap filling) in 4.0 under-performs (being polite here). My alternative implementation can do much better (YMMV). If you do try mine and it doesn't work OK, please let me know. Thanks.
https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

BagelOrb, 2019-03-18:
I am saying that thin walls in 4.0-beta works as normal, but it is 100% broken in 4.0-beta2

smartavionics, 2019-03-18:
I'm still saying that the thin wall (and gap filling support) in 4.0 under-performs.

BagelOrb, 2019-03-18:
Something seems to be wrong in your build, Mark:
```[Backend] [ERROR] Trying to retrieve setting with no value given: 'cool_fan_min_layer'```

BagelOrb, 2019-03-18:
I am planning on fixing the thin walls within the next year integrally.
I hope my solution will outperform even yours.

I remember we already talked about your gap filling algorithms somewhere. Do you remember which thread?

smartavionics, 2019-03-18:
Are you using the AppImage or just the CuraEngine?

BagelOrb, 2019-03-18:
Cura-mb-master-20190202.AppImage

smartavionics, 2019-03-18:
Hmm, I don't know why that setting is not found but you should try a much more recent release.

BagelOrb, 2019-03-18:
Oh wait Ill retry with latest appimage
I tried an old one

BagelOrb, 2019-03-18:
Can you give a quick recap of the general idea of your implementation of gap filling?

BagelOrb, 2019-03-18:
[image]
Mark, your version has some 'minor issues' 
;)
Cura-mb-master-20190317.AppImage

smartavionics, 2019-03-18:
Sorry, I can't remember the issue where we discussed my implementation.

I don't know if I will ever submit it as a PR as I know that in some situations it will not produce nice output and also I think it's quite slow. but for most simple cases it works well.

So, for both thin walls and gap filling, I estimate the points that define the centre line of the region that needs to be filled and I also estimate how wide the region is at those points. I also calculate the area that is drawn so that when the lines are output it avoids drawing over the same region twice.

smartavionics, 2019-03-18:
What's happening in your image is that you have a combination of thin walls and fat (not thin) walls. Try increasing the normal wall line width so that all the walls are printed as thin walls.

BagelOrb, 2019-03-18:
Please compare with the images I just attached to the original issue post. There is some big problem from which Ultimaker Cura is not suffering.

BagelOrb, 2019-03-18:
[image]
This is what it should look like - a lithophane of my ultimaker forum logo.

smartavionics, 2019-03-18:
Could you attach the model please?

smartavionics, 2019-03-18:
Ah, I see you already have, ignore that request!

smartavionics, 2019-03-18:
That's weird, I just sliced your model and it looks OK...

[image]

BagelOrb, 2019-03-18:
I'm sorry. It had some obscure profile selected from ages ago. Indeed there's no problem in your version.

smartavionics, 2019-03-18:
Thin walls, fat walls and filled wall gaps, all in perfect harmony...

[image]

smartavionics, 2019-03-18:
I do think that the transitions from fat to thin walls are very tricky to print nicely.

BagelOrb, 2019-03-18:
Mark Burtons version:
[image]

4.0.0-beta:
[image]

smartavionics, 2019-03-18:
So why did my comments get zapped?

BagelOrb, 2019-03-18:
Just cleaning up conversation which is irrelevant to this issue.
My misuse of your version of Cura is to no use for anybody.

smartavionics, 2019-03-18:
OK, just worried that it was the thought police erasing my history!

Ghostkeeper, 2019-03-18:
Maybe caused by [this](https://github.com/Ultimaker/CuraEngine/commit/9b4097d09ad4528a80097d0b900970c18b02f17c) or [this](https://github.com/Ultimaker/CuraEngine/commit/1e1e72d905cf6e9f42d85d3a8fc1d603e57c2b44)?

BagelOrb, 2019-03-18:
The first one is unlikely the culprit. The second one is a more likely culprit.

smartavionics, 2019-03-18:
It will definitely be related to the changes they made to MergeInfillLines as that is what is meant to turn the infill lines into beautiful smooth curves.

BagelOrb, 2019-03-18:
So perhaps it was this one: https://github.com/Ultimaker/CuraEngine/commit/ca9e4ae581658a33a7265aee12e2f7d5f26eea63#diff-cd4c69d037508b392e5eff0749d08c45

Ghostkeeper, 2019-03-18:
That commit was made before the first beta was released though.

BagelOrb, 2019-03-18:
Ok well, then we should just do a bisect!

Ghostkeeper, 2019-03-18:
One of our QA is doing that right now, using daily builds. Problem is though: The bug reproduces on January 8th, but not on December 18th (the date of the 1st beta). But there have been no commits on CuraEngine in the 4.0 branch in that period.

smartavionics, 2019-03-18:
Note that the hack I inserted to disable the path optimization when gyroid infill was used wasn't removed until 23rd Jan.

smartavionics, 2019-03-18:
Although, Tim's profile isn't using gyroid infill anyway.

BagelOrb, 2019-03-18:
Maybe the merge was in that period, while the commit was from earlier ;)

BagelOrb, 2019-03-18:
Or maybe a magic evil ghost creeped into the build system.
We all know you tend to keep those around ;)

Ghostkeeper, 2019-03-18:
We found it. It's this one: https://github.com/Ultimaker/CuraEngine/commit/6523774aa453e9dd2476991a671242fb6ba75182

That one was merged on the 18th (the day of the release of the first beta) but after we built the beta release, so we were confused by that.

We also discussed it (with the history in CURA-5776). I actually also found this issue and raised a ticket because there are certain thicknesses of parallel walls where this happens. We can't control all cases using the current method. We have two tickets open about this:
* One to split up the cases of thin walls from the cases of surfaces ending in a corner, a longer-term fix.
* One to tune the current algorithm better so that it at least works correctly for all parallel walls (like this case).

Neither have made it into 4.0 though.

smartavionics, 2019-03-18:
https://github.com/Ultimaker/CuraEngine/commit/6523774aa453e9dd2476991a671242fb6ba75182 contains changes in lots of different files but none are in MergeInfilLines so which of those changes is the culprit?

BagelOrb, 2019-03-18:
That's a merge commit! Are you saying the merge went wrong or that a bad commit was merged?

Anyway I think the least we should do is to revert the changes which were the culprit, because the intended fix was less bad than the currently broken feature.

BagelOrb, 2019-03-18:
I skipped through the merge commit changes, but couldn't find the mistake...

smartavionics, 2019-03-18:
But you can't revert that merge commit as it would trash loads of useful changes.

BagelOrb, 2019-03-18:
@smartavionics Nice winter design on the top banner, btw ;)

nallath, 2019-03-18:
The commit that caused the behavior was merged since the merge commit. Reverting it would fix this issue but it would re-introduce others.

BagelOrb, 2019-03-18:
Then revert it and fix those others.

Ghostkeeper, 2019-03-18:
I'd rather work in the current direction than to revert the 20 commits and re-do essentially the same fixes. As SmartAvionics said, it would throw quite a lot of useful changes away.

We already have three open issues to fix this: The two I mentioned [before](https://github.com/Ultimaker/Cura/issues/5445#issuecomment-473919468) in this thread and Smartavionics' version. We intend to combine the three. That should be sufficient.

BagelOrb, 2019-03-19:
Oh that's even better.

As long as it gets fixed before the 4.0 release.

Ghostkeeper, 2019-03-19:
We will not be able to fix this before the 4.0 stable release (i.e. in about an hour).

BagelOrb, 2019-03-19:
I think it is not allowable to have this bug in a stable release.
I'm not putting any effort in trying to delay an unstoppable train, though.

D:'

Ghostkeeper, 2019-03-20:
Yeah, sorry. If it makes you feel any better, the problem was worse in version 3.6 :sweat_smile:

BagelOrb, 2019-03-20:
No, I've tested it to be working in 3.6.

It is 100% broken in 4.0.0-beta2, how could it be worse?
The gap filling now *always* consists of ragged lines - they are *never* combined into smooth lines.

BagelOrb, 2019-03-20:
Take a better look at my screenshots.

BagelOrb, 2019-03-20:
I thought my test case was pretty representative for most cases where people have models with thin walls, but maybe you think my case is an edge-case..?

BagelOrb, 2019-03-20:
I discussed it in person and we found that the specific case I present here is indeed not that representative for all thin walls / gap filling cases.

Now I don't consider this issue any different from the general issue that gap filling and thin walls are generally problematic.

Ghostkeeper, 2019-08-11:
We've found that disabling the Fill Small Gaps setting causes the lines from Thin Walls to be joined properly. That may be a workaround for some (but filling small gaps is necessary for this lithophane anyway).

LipuFei, 2019-08-14:
Fixed in https://github.com/Ultimaker/CuraEngine/pull/1105

closed 2019-08-15 by Vandrasc