Ultimaker/Cura #5447: Thin walls redux

FlailAway, 2019-03-18:
My previous thread is closed but I need to mention some more stuff. I hope I am not peeing on someone's foot here. :)

**Application Version**
3.6.0

**Platform**
Mint 19.1/Cinnamon

**Printer**
Any, probably

**Steps to Reproduce**
Slice with odd number of walls and infill width equal to an extra wall.

**Actual Results**
Wastes hours on 0.4mm infill strokes

**Expected results**
Just run an extra wall

**Additional Information**
First off, let me say that SmartAvionics Fork Appimage is pretty dang fine. Thin wall infill conversion to extra walls works perfectly. Using the same .obj file as before and the same print-Profile it has dropped the print time from 2h:9m, 1h:24m!. Awesome stuff Avionics-dude !!

The last thread was closed with a comment "gap filling" but I think that is incorrect as the "gap" is/was exactly 0.4mm, the nozzle width and also the current wall width, as SmartAvionic's Fork shows how to do it.

I have a few cosmetic suggestions for SmartAvionics, but how do I pass them on?

Thanks.

smartavionics, 2019-03-18:
Hello @FlailAway , please open an issue at https://github.com/smartavionics/Cura or https://github.com/smartavionics/CuraEngine to discus anything related to my Cura release. Thanks.

Ghostkeeper, 2019-03-19:
Closing this as a duplicate of #5445 (and the cosmetic suggestions can be placed in his fork then).

closed 2019-03-19 by Ghostkeeper