Ultimaker/Cura #5448: How to access settings in post-process plugin

regularlabs, 2019-03-18:
I'm trying to create my own version of the PauseAtZ (or FilamentChange) plugin, as the ones included in Cura all have issues.

All going well, except I am not sure how to access the values like:
- material_bed_temperature
- material_print_temperature
- fan_speed

These should of course be the values taken from the project settings.

Any idea how to access these values?
I tried things like:
```
Application.getInstance().getGlobalContainerStack().getProperty("material_bed_temperature", "value")
self.getSettingValueByKey("material_bed_temperature")
``` 
But not getting what I need...

Ghostkeeper, 2019-03-19:
> `self.getSettingValueByKey("material_bed_temperature")`

This won't work. You can get a setting from your post-processing script this way (the ones from the JSON data that you provide in your script) but not the normal slicing settings.

> `Application.getInstance().getGlobalContainerStack().getProperty("material_bed_temperature", "value")`

This should work for global settings. For per-extruder settings like `material_print_temperature` you'd need to specify which extruder you'd like to get the setting from though. For that I'd use this:

```
extruder_nr = 0
CuraApplication.getInstance().getExtruderManager().getActiveExtruderStacks()[extruder_nr].getProperty("material_print_temperature", "value")
```

regularlabs, 2019-03-21:
Ok, thanks :)

closed 2019-03-21 by regularlabs