Ultimaker/Cura #5455: Cura 4.0.0 Windows 10 Full Screen issue

mc-ott, 2019-03-19:
**Application Version**
 Version 4.0.0

**Platform**
 Windows 10

**Printer**
 Original Prusa MK3 and G-code generation.

**Steps to Reproduce**
 Open Cura 4 and Toggle full screen mode from the View menu

**Actual Results**
  you will loose all top menu bar controls, the screen becomes inactive. Only F11 will reduce the window to original size. 

**Expected results**
 I think top menu bar should be active or a message should tell what key combination to hit (when the mouse goes up to the top menu bar level) to get back to the original window size.  

**Additional Information**

Ellecross, 2019-03-19:
Hey @mc-ott thanks for reporting this. We are aware of this issue. Actually, we have a ticket open now for this. ;) 

(for Cura devs, CURA-5925)

Ghostkeeper, 2019-03-20:
Not able to reproduce this in Windows 8.