Ultimaker/Cura #5457: Cura 3.6 Bed temp changes from 100C to 60C during print

shaddow501, 2019-03-19:
Where from the cura pull the 60C value?
please look at the pictures.
[image]

But the settings are 100C (for ABS).

Why cura decide to take it to 60C??
it ruin the prints, the ABS getting loose from the bed.

[image]

closed 2019-03-19 by shaddow501

Ghostkeeper, 2019-03-20:
Did you close this because you figured out what the reason was?

Looks like an issue of global version vs. per-extruder version with resolving.

shaddow501, 2019-03-20:
Hello Ghostkeeper

I closed it because I have found out that the problem may not be from the cura slicing.
I use plugin that I can print from cura to the Repetier server.

When I send the print from cura to the repetier server I notice that during the printing (not all the time but does happen) the Bed heater is turned off and also noticed that the extruder heater turns off before the printing ends (could happen in the middle of the print)

When I slice to file and upload the  gcode file to the repetier server the printing goes well.
So it seems that the cura slice it correctly and the problem comes maybe from the Cura=>Repetier server plugin.

Than I closed this issue here and opened the issue at the plugin maker:
https://github.com/VMaxx/RepetierIntegration

Still I dont understand why in the profiles it shows 60C but it does not affect the bed heater.