Ultimaker/Cura #5458: Split model into parts disappeared

paparico, 2019-03-19:
**Application Version**
Cura 4.0

**Platform**
Linux (Kubuntu 18.04)

**Printer**
UMO+ (though irrelevant)

**Steps to Reproduce**
Load a multi parts model, try to split into parts (in order, for instance, to make it fit onto the plate)
Search for the good old "Split Model into parts"
do not find it
cry a litle

**Actual Results**
i cannot split the model into parts. 
when i move/rotate/scale the model, the 3 parts are affected

**Expected results**
3 parts can be moved and placed independently on the build plate

**Additional Information**
[image]

smartavionics, 2019-03-19:
I think the mesh tools plugin that is available in the marketplace can split meshes...

[image]

smartavionics, 2019-03-19:
Yes, it can, I just tested it.

Ghostkeeper, 2019-03-20:
Probably this just happened because of the version upgrade and the user needs to update their plug-ins before they work again. We need to improve that UX.

nallath, 2019-03-21:
The next version of Cura (4.1) is expected to have a notification system for plugin updates, so that should make it a lot easier to see that this is the case.

paparico, 2019-03-22:
Ok, i thouht this was a core feature, i did not realize that this functionnality was provided by a plugin.

closed 2019-03-22 by paparico