Ultimaker/Cura #5459: Cura 4.0 and Repetier-Server connected printers

betavr, 2019-03-20:
Excuse me if this has been already discussed, but I'm unable to find any information on it. Please point me to the right issue if this is the case.

Problem description: I installed Cura 4.0 over my correctly configured 3.6 instance with two Repetier-Server connected printers. Although the configuration for my Repetier-Server connected printers were imported correctly, they show up as local printers in Cura 4.0.

I tried to solve this by:

- Searching the Marketplace for a Repetier-Server connection plugin, but they are greyed out (they aren't compatible with Cura 4.0).
- Trying to find an option in the menu for setting up network printers, but was unable to locate it.

So the question is: is my approach wrong, or does Cura 4.0 not have Repetier-Server connection capability available at this point in time?

Ghostkeeper, 2019-03-20:
Yes, I'm afraid that the maintainer of that plug-in didn't update it yet for the new UI.

This is something that you might want to report here: [Repetier integration repository](https://github.com/VMaxx/RepetierIntegration/issues)

closed 2019-03-20 by Ghostkeeper

betavr, 2019-03-20:
Will do, thanks. Keep up the good work!

VMaxx, 2019-03-22:
I am just about to post it.  I just got it uploaded to github.