Ultimaker/Cura #5461: Support blocker create empty part where 2 meshes intersect

BenRQ, 2019-03-20:
**Application Version**
4.0.0-BETA-2

**Platform**
Windows 10

**Printer**
Alfawise U20
[CFFFP_MeshBody7.zip](https://github.com/Ultimaker/Cura/files/2988556/CFFFP_MeshBody7.zip)

**Steps to Reproduce**
1) Import a model with interlapping part

[image]

2) check that the model is full in the "Prepare" view.

[image]

3) Add a support blocker anywhere, and set it to "Modify setting for overlap with other model".

4) Make sure that "Union overlapping volumes" setting is set to true

5) Slice => Problem occurs

6) remove support blocker => Problem disappear

**Actual Results**
The part of the model overlapping with each other are missing.
In the example below, the bottom should be full, and not have empty part.
[image]

**Expected results**
The behaviour should be the same, whether you are using support blocker or not.

**Additional Information**
This issue was originally reported here as a question:
https://community.ultimaker.com/topic/27121-model-get-empty-part-after-slicing-where-2-meshes-intersect/
Please check in this forum thread for additional information if required.

Ghostkeeper, 2019-03-21:
Your model is not watertight. Around the text there are some extra walls inside, by the looks of it. You can see this in X-ray view when looking from the side.
[image]

When using the modifier mesh thing, the polygons get broken up into multiple parts (each with different settings). This could cause the issue to rise to the surface where it was previously hidden by being lucky with how polygons were nested.

Ghostkeeper, 2019-03-21:
I'm not able to reproduce this issue with overlapping models that are properly watertight.

closed 2019-03-21 by Ghostkeeper

BenRQ, 2019-03-22:
OK thank you for the investigation, and the explanation @Ghostkeeper.
I'll try to make my model watertight.