Ultimaker/Cura #5462: Scaling Z is not functioning in Non-Uniform mode

baznr, 2019-03-20:
**Application Version**

4.0.0
**Platform**

Windows-64
**Printer**

Lader 3
**Steps to Reproduce**

Scale
deselect UNIFORM SCALING
changing Y, changes also Z
Changing Z, changes also Y
...
**Actual Results**

Impossible to scale non uniformly
**Expected results**

The Z axis must be scaled per se

**Additional Information**

Ghostkeeper, 2019-03-21:
Did you happen to rotate the model first before scaling it nonuniformly? This looks like a duplicate of #4136 if you did.

baznr, 2019-03-21:
Is not a duplicate, because this referenced by me is for the v.4.0.0

In fact -I had not realized this before your suggestion- if the rotation
comes after the scaling, the problem does not appear!
In any case, the behavior of the SW is a bit strange...

But for now the important is that there is at least one solution to the
problem (I must scale the objects x:101.6% , y:102.5%, z:100% to get a
matching printing)

Best regards.

Ghostkeeper, 2019-03-22:
Then it is a duplicate. Yes, the same bug can happen across multiple releases if we haven't fixed it yet.

closed 2019-03-22 by Ghostkeeper

baznr, 2019-03-22:
You know better sir!
In any case seems SO SIMPLE to fix that!
It's enough to DON'T call the recalculation subroutine.

Ghostkeeper, 2019-03-26:
The problem is a bit more hairy. After making any transformation, be it rotation, scaling, mirroring or moving, we're calculating what the current scale percentages are. There is probably a bug somewhere there where the scale percentages are using the wrong original width (untransformed instead of transformed or vice-versa).

This problem is very complex and open to subjective interpretation as well. For instance, if you rotate a mesh 45 degrees around the Z axis, then scale the X axis to 200% and then rotate the mesh 45 degrees back, what should the X and Y scale percentage look like?

If we were to not recalculate some of the axes then there would be a big jump in the percentage as soon as you so much as breathe on one of the axes that does require a recalculation.

baznr, 2019-03-27:
In my case, I do scaling because I found that (probably for mechanics
limitation) a circle in the plane XY is an light oval in the X axis.
And I think that the non-uniform scaling is requested for this purpose
mainly.

So, there is no need to rotate the non-uniformly scaled axis, because is
NOT a propriety of the rotated object, but instead OF THE AXIS shelf!

Un other solution could be to create a different setting, something like:
"AXIS SCALING/CALIBRATION", that could be even more useful and elegant.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Ghostkeeper, 2019-03-28:
Until those bugs are fixed, I advise you not to look at the percentages since they are wrong. Also take a good hard look at your firmware to see if the steps per millimetre is calibrated correctly.